jacksonliam/mjpg-streamer #163: INPUT_UVC: Allow framerates lower than that of the YUV camera

lhartmann, 2018-09-07:
**Problem:**
Input UVC framerate seems limited to the camera's driver suported options, even in YUV cameras.

**Steps to Reproduce:**
Request 120Hz or 1Hz from a camera that only supports 30Hz.

**What SHOULD happen:**
120Hz should be coerced to 30Hz, for it does not make sense.
1Hz should be accepted normally.

**What ACTUALLY happens:**
Both 1Hz and 120Hz are coerced to 30Hz, causing problems on weak CPU/network systems.

**P.S.1:** If the camera driver does not support 1Hz, but only 30Hz, why not just drop 29 out of 30 frames without processing? This should work for any integer divisor of the camera's framerate.

**P.S.2:** I do not know how MPG cameras work, so I believe it may not be so easy on those. That's why I mentioned YUV cameras only.

jacksonliam, 2018-09-08:
Lots of people were having troubles with the softfps code. And with it there was no control over what fps to run the camera at. E.g if you wanted 60 dropped to 10, but the camera had a 15 fps mode, it might have  choosen that.

It's now under the -softfps argument (provide the fps argument as the one you want the camera to run at and softfps as the framedroped result you want).

You can provide them the same (e.g 1fps) and the camera driver will choose what fps to run at if it doesn't support 1 fps.

Leaving this issue open to remind me to update the documentation.

dstulken, 2019-08-25:
@jacksonliam 
>Leaving this issue open to remind me to update the documentation.

Friendly reminder to update the documentation.  👍 

This is a fantastic and *extremely* useful feature - one that helps me very much, but I never would have known about had I not come here to post about a different issue.  

Some quick testing of some cams in YUV mode, one 640x480@15fps camera saw its CPU usage drop from 23% to ~7.5% with a softfps of 5.   Another at 720x480@10fps dropped from 42% to 21% CPU with a softfps of 5.   This makes a *huge* difference for systems running multiple cameras with lower FPS demands than what the cameras natively support, basically any system that tops out on either CPU cycles or network bandwidth.

I would imagine that quite a few other people would love to use this feature... if they knew about it!  :)