jacksonliam/mjpg-streamer #172: Raspbian GNU/Linux 9 (stretch) installation problem.

yuyat, 2018-10-23:
`# apt-get update 
# apt-get upgrade

sudo apt-get install cmake libjpeg8-dev
sudo apt-get install gcc g++

git clone https://github.com/jacksonliam/mjpg-streamer
cd mjpg-streamer/mjpg-streamer-experimental
make

# make
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMAKE_BUILD_TYPE = Release
-- Looking for include file sys/inotify.h
-- Looking for include file sys/inotify.h - found
CMake Warning at plugins/input_opencv/CMakeLists.txt:4 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'libgphoto2'
--   No package 'libgphoto2' found
-- Could NOT find GPHOTO2 (missing:  GPHOTO2_LIBRARY GPHOTO2_INCLUDE_DIR) 
-- Looking for include file linux/videodev2.h
-- Looking for include file linux/videodev2.h - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Could NOT find SDL (missing:  SDL_LIBRARY SDL_INCLUDE_DIR) 
-- Could NOT find protobuf-c (missing:  PROTOBUF_C_LIBRARY PROTOBUF_C_INCLUDE_DIR) 
-- 
-- The following features have been enabled:

 * PLUGIN_INPUT_FILE , File input plugin
 * PLUGIN_INPUT_HTTP , HTTP input proxy plugin
 * PLUGIN_INPUT_RASPICAM , Raspberry Pi input camera plugin
 * PLUGIN_INPUT_UVC , Video 4 Linux input plugin
 * PLUGIN_OUTPUT_FILE , File output plugin
 * PLUGIN_OUTPUT_HTTP , HTTP server output plugin
 * PLUGIN_OUTPUT_RTSP , RTSP output plugin
 * PLUGIN_OUTPUT_UDP , UDP output stream plugin

-- The following OPTIONAL packages have been found:

 * Threads

-- The following features have been disabled:

 * WXP_COMPAT , Enable compatibility with WebcamXP
 * PLUGIN_INPUT_OPENCV , OpenCV input plugin (unmet dependencies)
 * PLUGIN_INPUT_PTP2 , PTP2 input plugin (unmet dependencies)
 * ENABLE_HTTP_MANAGEMENT , Enable experimental HTTP management option
 * PLUGIN_OUTPUT_VIEWER , SDL output viewer plugin (unmet dependencies)
 * PLUGIN_OUTPUT_ZMQSERVER , ZMQ Server output plugin (unmet dependencies)

-- The following OPTIONAL packages have not been found:

 * OpenCV
 * Gphoto2
 * SDL

-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/mjpg-streamer/mjpg-streamer-experimental/_build
make -C _build
make[1]: se entra en el directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
make[2]: se entra en el directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: se entra en el directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
Scanning dependencies of target mjpg_streamer
make[3]: se sale del directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: se entra en el directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
[  4%] Building C object CMakeFiles/mjpg_streamer.dir/mjpg_streamer.c.o
[  8%] Building C object CMakeFiles/mjpg_streamer.dir/utils.c.o
[ 12%] Linking C executable mjpg_streamer
make[3]: se sale del directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 12%] Built target mjpg_streamer
make[3]: se entra en el directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
Scanning dependencies of target input_file
make[3]: se sale del directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: se entra en el directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 16%] Building C object plugins/input_file/CMakeFiles/input_file.dir/input_file.c.o
[ 20%] Linking C shared library input_file.so
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libc.a(dl-tsd.o)(.text+0x14): la reubicación R_ARM_TLS_LE32 no se permite en objetos compartidos
/usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libc.a(dl-tsd.o): En la función `_dl_initial_error_catch_tsd':
(.text+0x14): reubicación peligrosa: no se admite la reubicación
collect2: error: ld returned 1 exit status
plugins/input_file/CMakeFiles/input_file.dir/build.make:94: fallo en las instrucciones para el objetivo 'plugins/input_file/input_file.so'
make[3]: *** [plugins/input_file/input_file.so] Error 1
make[3]: se sale del directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
CMakeFiles/Makefile2:122: fallo en las instrucciones para el objetivo 'plugins/input_file/CMakeFiles/input_file.dir/all'
make[2]: *** [plugins/input_file/CMakeFiles/input_file.dir/all] Error 2
make[2]: se sale del directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:127: fallo en las instrucciones para el objetivo 'all'
make[1]: *** [all] Error 2
make[1]: se sale del directorio '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:17: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error 2
`
=========

`cat  /etc/os-release

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

yuyat, 2018-10-23:
> Any chance you can get make to run in English? I cannot understand the error!
Sure, my fault.
I tried to downgrade gcc version to make sure that version 6.3 doesn't have bugs. (ocurrs the same)
In raspbian Jessie + GCC 4.9.2 it works perfect.
```

[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
make -C _build
make[1]: Entering directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
make[2]: Entering directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Leaving directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
[  4%] Building C object CMakeFiles/mjpg_streamer.dir/mjpg_streamer.c.o
[  8%] Building C object CMakeFiles/mjpg_streamer.dir/utils.c.o
[ 12%] Linking C executable mjpg_streamer
make[3]: Leaving directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 12%] Built target mjpg_streamer
make[3]: Entering directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Leaving directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 16%] Building C object plugins/input_file/CMakeFiles/input_file.dir/input_file.c.o
[ 20%] Linking C shared library input_file.so
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/libc.a(dl-tsd.o)(.text+0x14): R_ARM_TLS_LE32 relocation not permitted in shared object
/usr/lib/gcc/arm-linux-gnueabihf/4.9/../../../arm-linux-gnueabihf/libc.a(dl-tsd.o): In function `_dl_initial_error_catch_tsd':
(.text+0x14): dangerous relocation: unsupported relocation
collect2: error: ld returned 1 exit status
plugins/input_file/CMakeFiles/input_file.dir/build.make:94: recipe for target 'plugins/input_file/input_file.so' failed
make[3]: *** [plugins/input_file/input_file.so] Error 1
make[3]: Leaving directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
CMakeFiles/Makefile2:122: recipe for target 'plugins/input_file/CMakeFiles/input_file.dir/all' failed
make[2]: *** [plugins/input_file/CMakeFiles/input_file.dir/all] Error 2
make[2]: Leaving directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:127: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/tmp/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:17: recipe for target 'all' failed
make: *** [all] Error 2

```

jacksonliam, 2018-10-25:
I have tried the latest stretch image on both a 3B and original B and cannot reproduce this. Which pi are you using and is there anything special about your setup e.g. what's been installed/configured that could be causing it? 

Have you done anything which could have changed libc, build paths, etc?

I'm not ruling it out being a linker flag issue or something I can fix, but It doesn't look like it happens on a 'clean' raspbian and I need to be able to reproduce it to fix it.

yuyat, 2018-10-25:
> I have tried the latest stretch image on both a 3B and original B and cannot reproduce this. Which pi are you using and is there anything special about your setup e.g. what's been [...]

3B Formatted a couple of days ago. Firmware updated with:
`rpi-update`

Installed:
```
nvm
Octoprint (as a service)
Openvpn
and dependencies
```
I'll try to format and start again.
Thank you