jacksonliam/mjpg-streamer #173: Input from stdin

plfort, 2018-10-24:
Hello,
Is it possible to process stdin with mjpg-streamer ?
I would like to do  : 
`gphoto2 --capture-movie --stdout | mjpg-streamer ...`

I saw the plugin input_ptp2 but I can't make it work, I get this error : 
`mjpg_streamer: symbol lookup error: /usr/local/lib/mjpg-streamer/input_ptp2.so: undefined symbol: gp_context_new`

jacksonliam, 2018-10-24:
Hey, we don't support input from stdin,
However,
I did fix the link error for input_ptp2 - I guess no-one has tried to use it in years!

Do a 
``
make distclean
git pull
make
``
And it should fix the `undefined symbol: gp_context_new` error

plfort, 2018-10-24:
Thank you !
It is better : the camera is detected and the capture seems to start (mirror is raised) but finish with a Segmentation fault.
I'm not a C developer but I can try ... I'll make a PR if I can solve the problem.

jacksonliam, 2018-10-25:
Ah that's a shame.

I don't have a compatible camera to fix it myself but if you can't find the bug then do a debug build (see below), run it and send me the corefile from the crash I might be able to work it out.

```
make distclean
make CMAKE_BUILD_TYPE=Debug
```

jacksonliam, 2018-10-25:
As a stab in the dark, there is no buffer overrun protection and whoever wrote the plugin assumed the camera wouldn't send frames above 256KB...

As a test, modify the 256 on this line to something more than big enough to hold what your camera can send, e.g. 10000. 
https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_ptp2/input_ptp2.c#L134

plfort, 2018-10-25:
Hey!
I stopped there yesterday after I raised it to 512, and it worked !
But I was thinking of making a better fix : maybe usining realloc to fit xsize ?

jacksonliam, 2018-10-25:
Good news! Yes that would be the better fix.  

If the buffer is too small realloc to something bigger than xsize plus some margin (10% ?), exiting gracefully if the realloc fails.