jacksonliam/mjpg-streamer #176: raspicam start failed

sman2013, 2018-10-31:
When I use `./mjpg_streamer -i "input_raspicam.so" -o "output_http.so -w /www"` to start mjpg, an error accur.
`mmal: mmal_component_create_core: could not find component 'vc.ril.camera'
error create camera`
But when I use `./mjpg_streamer -i "input_uvc.so" -o "output_http.so -w /www"`, it works well.
I don't know what's wrong with it.

sman2013, 2018-11-01:
@djkraven RaspberryPi camera

slimcdk, 2019-02-28:
Did you face the ribbon cable the correct way?
Did you put the ribbon cable in the correct slot on the RPi (not in display slot tho it fits)

mcr222, 2022-01-28:
https://www.raspberrypi.com/documentation/accessories/camera.html

In order to work with new Raspbian, you need to enable legacy camera:
https://www.raspberrypi.com/documentation/accessories/camera.html#re-enabling-the-legacy-stack