jacksonliam/mjpg-streamer #181: vcos.h missing?

JakedPotato, 2018-11-22:
got a fresh install of raspbian running and I keep on encountering this issue :(

`[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMAKE_BUILD_TYPE = Release
-- Looking for include file sys/inotify.h
-- Looking for include file sys/inotify.h - found
CMake Warning at plugins/input_opencv/CMakeLists.txt:4 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'libgphoto2'
--   No package 'libgphoto2' found
-- Could NOT find GPHOTO2 (missing:  GPHOTO2_LIBRARY GPHOTO2_INCLUDE_DIR) 
-- Looking for include file linux/videodev2.h
-- Looking for include file linux/videodev2.h - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Could NOT find SDL (missing:  SDL_LIBRARY SDL_INCLUDE_DIR) 
-- Could NOT find protobuf-c (missing:  PROTOBUF_C_LIBRARY PROTOBUF_C_INCLUDE_DIR) 
-- 
-- The following features have been enabled:

 * PLUGIN_INPUT_FILE , File input plugin
 * PLUGIN_INPUT_HTTP , HTTP input proxy plugin
 * PLUGIN_INPUT_RASPICAM , Raspberry Pi input camera plugin
 * PLUGIN_INPUT_UVC , Video 4 Linux input plugin
 * PLUGIN_OUTPUT_FILE , File output plugin
 * PLUGIN_OUTPUT_HTTP , HTTP server output plugin
 * PLUGIN_OUTPUT_RTSP , RTSP output plugin
 * PLUGIN_OUTPUT_UDP , UDP output stream plugin

-- The following OPTIONAL packages have been found:

 * Threads

-- The following features have been disabled:

 * WXP_COMPAT , Enable compatibility with WebcamXP
 * PLUGIN_INPUT_OPENCV , OpenCV input plugin (unmet dependencies)
 * PLUGIN_INPUT_PTP2 , PTP2 input plugin (unmet dependencies)
 * ENABLE_HTTP_MANAGEMENT , Enable experimental HTTP management option
 * PLUGIN_OUTPUT_VIEWER , SDL output viewer plugin (unmet dependencies)
 * PLUGIN_OUTPUT_ZMQSERVER , ZMQ Server output plugin (unmet dependencies)

-- The following OPTIONAL packages have not been found:

 * OpenCV
 * Gphoto2
 * SDL

-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/mjpg-streamer/mjpg-streamer-experimental/_build
make -C _build
make[1]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[2]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
Scanning dependencies of target mjpg_streamer
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[  4%] Building C object CMakeFiles/mjpg_streamer.dir/mjpg_streamer.c.o
[  8%] Building C object CMakeFiles/mjpg_streamer.dir/utils.c.o
[ 12%] Linking C executable mjpg_streamer
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 12%] Built target mjpg_streamer
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
Scanning dependencies of target input_file
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 16%] Building C object plugins/input_file/CMakeFiles/input_file.dir/input_file.c.o
[ 20%] Linking C shared library input_file.so
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 20%] Built target input_file
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
Scanning dependencies of target input_http
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 24%] Building C object plugins/input_http/CMakeFiles/input_http.dir/input_http.c.o
[ 28%] Building C object plugins/input_http/CMakeFiles/input_http.dir/misc.c.o
[ 32%] Building C object plugins/input_http/CMakeFiles/input_http.dir/mjpg-proxy.c.o
[ 36%] Linking C shared library input_http.so
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 36%] Built target input_http
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
Scanning dependencies of target input_raspicam
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 40%] Building C object plugins/input_raspicam/CMakeFiles/input_raspicam.dir/input_raspicam.c.o
In file included from /opt/vc/include/bcm_host.h:50:0,
                 from /home/pi/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/input_raspicam.c:44:
/opt/vc/include/interface/vmcs_host/vc_dispmanx.h:33:33: fatal error: interface/vcos/vcos.h: No such file or directory
 #include "interface/vcos/vcos.h"
                                 ^
compilation terminated.
plugins/input_raspicam/CMakeFiles/input_raspicam.dir/build.make:62: recipe for target 'plugins/input_raspicam/CMakeFiles/input_raspicam.dir/input_raspicam.c.o' failed
make[3]: *** [plugins/input_raspicam/CMakeFiles/input_raspicam.dir/input_raspicam.c.o] Error 1
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
CMakeFiles/Makefile2:250: recipe for target 'plugins/input_raspicam/CMakeFiles/input_raspicam.dir/all' failed
make[2]: *** [plugins/input_raspicam/CMakeFiles/input_raspicam.dir/all] Error 2
make[2]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:127: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:17: recipe for target 'all' failed
make: *** [all] Error 2
`

it seems that vcos.h doesn't exist or something like that. I tried to install Raspbian via NOOBS. Is that the wrong approach?

jacksonliam, 2018-11-23:
Hmm I've not seen this, the file should be in the image. Looks like a potential bug with the pi image rather than mjpg-streamer, but could be a cmake thing. 

Are you compiling on an actual pi?

Does vcos.h exist?

takada-s, 2022-02-08:
I also encountered this problem and found a workaround.

- Raspberry Pi ZeroW
- Raspbian Gnu/Linux 11

I found that there is no `/opt/vc` folder so `plugins/input_raspicam/CMakeLists.txt` judges `HAS_RASPI OFF`.
but i have `vcos.h` at `/usr/include/interface/vcos/vcos.h`.

so I edited the CMakeList.txt as:
```diff
diff --git a/mjpg-streamer-experimental/plugins/input_raspicam/CMakeLists.txt b/mjpg-streamer-experimental/plugins/input_raspicam/CMakeLists.txt
index bb645b2..ecc42df 100644
--- a/mjpg-streamer-experimental/plugins/input_raspicam/CMakeLists.txt
+++ b/mjpg-streamer-experimental/plugins/input_raspicam/CMakeLists.txt
@@ -1,5 +1,5 @@

-if (EXISTS /opt/vc/include)
+if (EXISTS /usr/include/interface/vcos)
     set(HAS_RASPI ON)
 else()
     set(HAS_RASPI OFF)
@@ -10,13 +10,12 @@ MJPG_STREAMER_PLUGIN_OPTION(input_raspicam "Raspberry Pi input camera plugin"

 if (PLUGIN_INPUT_RASPICAM)

-    include_directories(/opt/vc/include)
-    include_directories(/opt/vc/include/interface/vcos)
-    include_directories(/opt/vc/include/interface/vcos/pthreads)
-    include_directories(/opt/vc/include/interface/vmcs_host)
-    include_directories(/opt/vc/include/interface/vmcs_host/linux)
+    include_directories(/usr/include/interface/vcos)
+    include_directories(/usr/include/interface/vcos/pthreads)
+    include_directories(/usr/include/interface/vmcs_host)
+    include_directories(/usr/include/interface/vmcs_host/linux)

-    link_directories(/opt/vc/lib)
+    link_directories(/usr/lib)

     MJPG_STREAMER_PLUGIN_COMPILE(input_raspicam input_raspicam.c)
```

and got `input_raspicam.so` successfully.

I'm not sure this method is reasonable, it works on my rpi.