jacksonliam/mjpg-streamer #182: select timeout

cgpeter96, 2018-11-24:
Re

jacksonliam, 2018-11-26:
@mdevaev multiple cams definitely used to work before the new select() stuff, any idea what the issue might be?

jacksonliam, 2018-11-26:
Actually I can see you are using an invalid frame rate, I think it has to be a whole number and will parse that as 0, doing whatever your camera does when asked to do 0. If the framerate is below one every 5 seconds, you'll want to give the `-timeout` arg in seconds.

I think we could automatically increase the timeout if we have a low framerate though, as 5 sec will be too slow. It would probably also be useful if we could accept fractional framerates.

cgpeter96, 2018-11-30:
@mdevaev  thanks ,I will try it

flixman, 2019-07-21:
Hi. I'm having exactly the same issue: when starting

    /usr/local/bin/mjpg_streamer -i input_uvc.so -y -d /dev/video0 -q 80 -n -f 5 -r 640x480 -timeout 15 -o output_http.so -p 8080
    /usr/local/bin/mjpg_streamer -i input_uvc.so -y -d /dev/video2 -q 80 -n -f 5 -r 640x480 -timeout 15 -o output_http.so -p 8082

at the same time, after some seconds/minutes I always get the timeout (Note: by "some seconds" I mean something as short as 2 secs). If I start the one for /dev/video2 I runs fine until I start the other: then both get the timeout.

This makes me think it might be a problem not related to mjpg_streamer, but to linux itself?

mdevaev, 2019-07-21:
It's on Raspberry Pi?

flixman, 2019-07-22:
sorry, my bad: Yes, on an RPi 3B+.

mdevaev, 2019-07-23:
Maybe you're right and the problem somewhere in the kernel.
Try to use other mjpg program with your cameras, for example my https://github.com/pi-kvm/ustreamer. If the problem persists, then it is probably in the kernel.

flixman, 2019-07-23:
It definitely seems to be a system problem, rather than app-related. I have coded a simple mjpeg streamer variant, using restbed to serve the images, and I'm experiencing exactly the same issue: At some point in time, a select starts to timeout, and from that moment all selects will fail. As soon as I kill and restart the app, everything gets fixed.

lbr38, 2019-09-01:
Hi,

Same issue here : 
```
 i: select() timeout
 i: cleaning up resources allocated by input thread
```
Running on Raspbian 10 (kernel 4.19.66-v7+) on RPI3

Can someone help us find what is the problem please ?

mdevaev, 2019-09-01:
Have you tried increasing the timeout?
`-timeout 15`

cgpeter96, 2019-09-02:
> Have you tried increasing the timeout?
> `-timeout 15`

no,  I just reduce the number of camera

lbr38, 2019-09-03:
Tried with -timeout 15 with no success, still the same issue.
Also tried to change the settings of the uvcvideo driver as suggested here : https://www.raspberrypi.org/forums/viewtopic.php?t=35689

> modprobe uvcvideo nodrop=1 timeout=5000 quirks=0x80

but no success either :(

mdevaev, 2019-09-03:
Does your camera work with another program?

cgpeter96, 2019-09-04:
no I just use another  strategy that we use `fswebcam`, because I just wanna take some photos every an hour.  It's not required real-time
```
# execute this command every an hour
fswebcam --no-banner -d /dev/video0 -r 1920x1080 -F 15  --save {1..15}.jpg 
```

cgpeter96, 2019-09-04:
> Does your camera work with another program

yeap,  it's work well on PC(x86-64)

mdevaev, 2019-09-04:
@cgpeter96, @lbr38 - Sorry, I'm a little confused. Do you solve your common problem? :)

Be that as it may, more diagnostics are needed. It is necessary to rebuild mjpg-streamer in debug mode and post here the entire log of the program.

Or there is an easier way. Try to use my https://github.com/pikvm/ustreamer, it's very similar to mjpg-streamer but contains many debug hints out of box: `ustreamer --debug --device /dev/video0`. Run it and show me the log. This is not advertising - I suffered from debugging of v4l2 and made my own tool to simplify this task.

lbr38, 2019-09-04:
> @cgpeter96, @lbr38 - Sorry, I'm a little confused. Do you solve your common problem? :)
>
> Be that as it may, more diagnostics are needed. It is necessary to rebuild mjpg-streamer in debug mode and [...]

Thank you for your help, in my case the problem is not solved (still got timeout error) and I'd like to try ustreamer to get more logs but I have issues while building it (i'm on Raspberry Pi3) :

```
$ make
-- CC src/http/mime.c
src/http/mime.c:27:10: fatal error: event2/util.h: Aucun fichier ou dossier de ce type (traduction : no file found)
 #include <event2/util.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:72: src/http/mime.o] Error 1
```

mdevaev, 2019-09-04:
Install libevent, libjpeg and libuuid

lbr38, 2019-09-04:
Can you tell me the equivalent packages names for RPi (debian) ? 
I tried with libevent-dev libjpeg-dev and uuid-dev but still got error while building ustreamer. I may have installed the wrong packages ?

```
sudo make
-- LINKING ustreamer
===== Build complete =====
== CC      = cc
== LIBS    = -lm -ljpeg -pthread -levent -levent_pthreads -luuid
== CFLAGS  = -O3 -c -std=c11 -Wall -Wextra -D_GNU_SOURCE
== LDFLAGS = 
/usr/bin/ld: src/encoder.o: in function `encoder_destroy':
encoder.c:(.text+0xec): undefined reference to `omx_encoder_destroy'
/usr/bin/ld: src/encoder.o: in function `encoder_prepare':
encoder.c:(.text+0x4d8): undefined reference to `omx_encoder_init'
/usr/bin/ld: encoder.c:(.text+0x86c): undefined reference to `omx_encoder_prepare'
/usr/bin/ld: src/encoder.o: in function `encoder_compress_buffer':
encoder.c:(.text+0xde4): undefined reference to `omx_encoder_compress_buffer'
collect2: error: ld returned 1 exit status
make: *** [Makefile:62: ustreamer] Error 1
```

Thank you

mdevaev, 2019-09-04:
Should be fine. run `make clean` and `make` again

lbr38, 2019-09-04:
Thanks, so here are some logs from ustreamer after 5sec of execution. I don't know if this will help.
The thing is, timeout occurs randomly with mjpg_streamer, it can be after minutes or after 1hour of execution.. ustreamer will produce so much logs if I run it for so long. 
Do you need all the logs to debug ? 

```
./ustreamer --device=/dev/video0  -r 1920x1080 -f 30 -m JPEG --debug
-- INFO  [293500.212 tid=419] -- Installing SIGINT handler ...
-- INFO  [293500.212 tid=419] -- Installing SIGTERM handler ...
-- INFO  [293500.212 tid=419] -- Ignoring SIGPIPE ...
-- INFO  [293500.213 tid=419] -- Using internal blank placeholder
-- DEBUG [293500.213 tid=419] -- Binding HTTP to [127.0.0.1]:8080 ...
-- INFO  [293500.213 tid=419] -- Listening HTTP on [127.0.0.1]:8080
-- INFO  [293500.214 tid=420] -- Using V4L2 device: /dev/video0
-- INFO  [293500.214 tid=421] -- Starting HTTP eventloop ...
-- INFO  [293500.214 tid=420] -- Using desired FPS: 30
-- DEBUG [293500.214 tid=420] -- _stream_init_loop: stream->proc->stop=0
================================================================================
-- DEBUG [293500.231 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.231 tid=421] -- HTTP: dropped same frame (BLANK) number 0
-- DEBUG [293500.249 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.250 tid=421] -- HTTP: dropped same frame (BLANK) number 1
-- DEBUG [293500.260 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.260 tid=421] -- HTTP: dropped same frame (BLANK) number 2
-- DEBUG [293500.277 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.277 tid=421] -- HTTP: dropped same frame (BLANK) number 3
-- INFO  [293500.297 tid=420] -- Device fd=8 opened
-- DEBUG [293500.298 tid=420] -- Calling ioctl(VIDIOC_QUERYCAP) ...
-- INFO  [293500.298 tid=420] -- Using input channel: 0
-- INFO  [293500.298 tid=420] -- Using TV standard: DEFAULT
-- DEBUG [293500.298 tid=420] -- Calling ioctl(VIDIOC_S_FMT) ...
-- DEBUG [293500.299 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.299 tid=421] -- HTTP: dropped same frame (BLANK) number 4
-- DEBUG [293500.309 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.309 tid=421] -- HTTP: dropped same frame (BLANK) number 5
-- DEBUG [293500.327 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.327 tid=421] -- HTTP: dropped same frame (BLANK) number 6
-- DEBUG [293500.349 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.349 tid=421] -- HTTP: dropped same frame (BLANK) number 7
-- INFO  [293500.352 tid=420] -- Using resolution: 1920x1080
-- INFO  [293500.352 tid=420] -- Using pixelformat: JPEG
-- DEBUG [293500.352 tid=420] -- Calling ioctl(VIDIOC_G_PARM) ...
-- DEBUG [293500.359 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.359 tid=421] -- HTTP: dropped same frame (BLANK) number 8
-- DEBUG [293500.376 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.376 tid=421] -- HTTP: dropped same frame (BLANK) number 9
-- DEBUG [293500.399 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- DEBUG [293500.409 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.409 tid=421] -- HTTP: dropped same frame (BLANK) number 0
-- DEBUG [293500.433 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.433 tid=421] -- HTTP: dropped same frame (BLANK) number 1
-- DEBUG [293500.447 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.447 tid=421] -- HTTP: dropped same frame (BLANK) number 2
-- DEBUG [293500.457 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.457 tid=421] -- HTTP: dropped same frame (BLANK) number 3
-- DEBUG [293500.481 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.481 tid=421] -- HTTP: dropped same frame (BLANK) number 4
-- DEBUG [293500.500 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.500 tid=421] -- HTTP: dropped same frame (BLANK) number 5
-- DEBUG [293500.510 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.510 tid=421] -- HTTP: dropped same frame (BLANK) number 6
-- DEBUG [293500.527 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.527 tid=421] -- HTTP: dropped same frame (BLANK) number 7
-- DEBUG [293500.549 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.549 tid=421] -- HTTP: dropped same frame (BLANK) number 8
-- DEBUG [293500.560 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.560 tid=421] -- HTTP: dropped same frame (BLANK) number 9
-- DEBUG [293500.577 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- DEBUG [293500.599 tid=421] -- Refreshing HTTP exposed (BLANK) ...
-- PERF  [293500.599 tid=421] -- HTTP: dropped same frame (BLANK) number 0
-- DEBUG [293500.609 tid=421] -- Refreshing HTTP exposed (BLANK) ...
[... 474 lines cut ...]
video capture
    VIDIOC_ENUM_FMT(0,VIDEO_CAPTURE)
	index                   : 0
	type                    : VIDEO_CAPTURE
	flags                   : 1
	description             : "Motion-JPEG"
	pixelformat             : 0x47504a4d [MJPG]
    VIDIOC_ENUM_FMT(1,VIDEO_CAPTURE)
	index                   : 1
	type                    : VIDEO_CAPTURE
	flags                   : 0
	description             : "YUYV 4:2:2"
	pixelformat             : 0x56595559 [YUYV]
    VIDIOC_G_FMT(VIDEO_CAPTURE)
	type                    : VIDEO_CAPTURE
	fmt.pix.width           : 1920
	fmt.pix.height          : 1080
	fmt.pix.pixelformat     : 0x47504a4d [MJPG]
	fmt.pix.field           : NONE
	fmt.pix.bytesperline    : 0
	fmt.pix.sizeimage       : 4147789
	fmt.pix.colorspace      : SRGB
	fmt.pix.priv            : 4276996862

```

mdevaev, 2019-09-04:
> timeout occurs randomly

It changes everything!

@jacksonliam I have an idea. In some cases, a timeout is not a fatal failure. You could add an option that would enable behavior in which the device will be reinitialized during a timeout or other error. This is not the fault of my patch. I investigated this question - without my patch mjpg_streamer simplye hangs in this case.

@lbr38 ustreamer correctly handles such situations and automatically restarts the stream on error or timeout. You can use it for your purposes if you want. If you have problems or questions, you can create an issue in my repository.

lbr38, 2019-09-04:
Okay I guess I will use ustreamer as you suggested, but will need some help to set up the http server. I'll open a new issue in your repository. 

Thank you for all

jacksonliam, 2019-09-04:
> It changes everything!
>
> @jacksonliam I have an idea. In some cases, a timeout is not a fatal failure. You could add an option that would enable behavior in which the device will be reinitialized [...]

Thanks, a ll think about making a change in this area, do you have to re-init everything or just carry on with more selects?

mdevaev, 2019-09-04:
I think you can re-init only v4l2 code. I.e. close device, open again and etc.

https://github.com/pikvm/ustreamer/blob/master/src/stream.c#L128

dstulken, 2019-09-11:
> Thanks, a ll think about making a change in this area

@jacksonliam, in a general sense, you might also consider a second parallel option to have mjpg-streamer terminate on errors like this.  From a user perspective, this select() timeout is a frustrating issue, as the server side happily continues running... so camera systems silently fail, yet the host doesn't have the ability to recover.   If mjpg-streamer had exited on the failure, it would have been possible to wrap the command in a script to auto-restart, providing an additional layer of fault recovery possibility.

That's not to say that this issue shouldn't be patched by restarting the input component within mjpg-streamer, as that is clearly the better solution.   But for general runtime faults with the camera inputs, continuing to run isn't necessarily always helpful, even if it is possible to continue.

Thanks for looking into this!

madscientist159, 2020-01-12:
@jacksonliam Thanks for the pointer to ustreamer!  So far it's completely stopped our constant random streamer lockups on multi-camera systems.

koba1995, 2020-08-19:
I also have similar problem. I'm using raspberrypi 2B+ and two same type cameras. I added systemd script for executing on boot and restart on failure.
Every time, 1 of 2 fails("select() timeout" message and no stream output), but the program keep running, so it's not restarted. Manual restart always succeeds.
The root cause doesn't seem to be in mjpg_streamer.

However, there is a problem that mjpg_streamer keeps running after some error occured in input_uvc.
I tryed fixing only this point.
b68606bfdd10fcdbcc05d9aa65b0b6101c900f18

ddacunha, 2021-06-13:
I had a `select() timeout` on my Raspberry Pi 4 until I increased the `gpu_mem` from `gpu_mem=128` to `gpu_mem=256` in `/boot/config.txt`.

reversefold, 2022-01-22:
Having the same problem. I have two webcams (one Logitech, one Razer). If I have both plugged into the same Raspberry Pi then either one will work fine if mjpg-streamer is started first for that camera. When mjpg-streamer is started for the second camera I never get an image and it ends up in select() timeout.

I have tried increasing gpu_mem to 256 with no effect. I have tried increasing the timeout with no effect. I have tried decreasing the frame rate and resolution with no effect. The first started always works and the second started always times out, with the first still working fine.

jacksonliam, 2022-01-23:
Definitely got enough power for the pi and both cameras? 

Assume this is running multiple instances of mjpg-streamer? Is the issue the same if you run two input plugins under one mjpg steamer?

Select timeouts is usually the OS / v4l2 not returning a frame, it's not usually an mjpg-streamer issue.

closed 2022-02-22 by cgpeter96

James4Ever0, 2024-09-08:
https://github.com/pikvm/ustreamer/issues/284

James4Ever0, 2024-09-08:
Running `dmesg | egrep 'tc35|1-1.4|uvc'` I get:

```
uvcvideo 1-1.5:1.1: Non-zero status (-71) in video completion handler.
```

James4Ever0, 2024-09-08:
This is not true. I use OBS and everything is fine. After switched to `ustreamer` or `mjpeg-streamer` everything is a mess. All I can say is that OBS handles some magic while all other libraries are just doing wrong.

James4Ever0, 2024-09-08:
I am afraid I have found the solution. Turned out pretty simple. It is to reset the USB device every time the line `CAP: Device select() timeout` showed up. One can write a script to achieve this if running under Docker.