jacksonliam/mjpg-streamer #183: 400: Not Found! No file extension found

dupeljan, 2018-11-29:
Hello! 
I tried to stream my webcam, typing
`mjpg_streamer -i "input_uvc.so -d /dev/video0 -y -f 2 -n -r 640x480" -o "output_http.so -p 7010 -w ./www"
MJPG Streamer Version: git rev: ddb69b7b4f114f3c2ca01adf55712792ca8aed43
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 2
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
 i: FPS coerced ......: from 2 to 5
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 7010
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled

but when I had followed link 
http://127.0.0.1:7010/stream 
I found this 
`400: Not Found!
No file extension found`

What'm I doing wrong?
my os: Ubuntu 18.04.1 LTS
mjpg-streamer installed successful by :
`cd mjpg-streamer-experimental
make
sudo make install`

jacksonliam, 2018-11-29:
`/stream` is not the correct link?

closed 2018-11-29 by jacksonliam

dupeljan, 2018-11-29:
Thank you so much!