jacksonliam/mjpg-streamer #184: What is the difference between the stream formats?

RobLewis, 2018-12-04:
I can't find any documentation of the difference between the various streams, like `stream_simple.html` and `javascript_simple.html` (and others). Can you explain? I have noticed that `javascript_simple.html` seems to load the CPU less than `stream_simple.html`. Explanation appreciated.

RobLewis, 2018-12-04:
Additionally, is there any way to detect when a stream has crashed or otherwise gone offline?

jacksonliam, 2018-12-04:
I think the html source is the documentation :-D

It's just different ways of showing the stream really.

It depends what you mean by crashed?surely you can detect it going offline because the web server is no longer responding?

closed 2018-12-04 by jacksonliam

RobLewis, 2018-12-04:
@jacksonliam Did you mean the C source? Where's the html source? 

I have 2 different cameras displayed in separate iframes on a page. I have a "watchdog" javascript that runs every few seconds. I'd like it to be able to detect that there is a problem with a stream and re-launch the streamer. 

Am I right that there are no `man` pages for the program?

jacksonliam, 2018-12-04:
You can see what those two different html pages are doing by looking at their source, they are very simple.

https://github.com/jacksonliam/mjpg-streamer/tree/master/mjpg-streamer-experimental/www

The documentation that exists is mainly in the Readme.md files.

https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/output_http/README.md

Cheers,
Liam.