jacksonliam/mjpg-streamer #185: output_file documentation?

eirhardt, 2018-12-10:
I'm looking for any and all info related to the output_file plugin. I already have two cameras successfully streaming to a webpage I designed (via `output_http.so`) but I'm hoping that there's a way I can also write what these cameras stream to a file. Is that possible?

jacksonliam, 2018-12-18:
Yes you can specify multiple output plugins (multiple -o options).

The help should be enough to get you going https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/output_file/output_file.c#L63

closed 2018-12-18 by jacksonliam

eirhardt, 2018-12-18:
I've actually been reviewing the help commands for `output_file.so` but I haven't gotten my command to work yet. 

Here's what I've been trying: 
`mjpg_streamer -i "/.../mjpg-streamer/input_uvc.so -n -f 30 -r 640x480 -d /dev/video0" -o "/.../mjpg-streamer/output_http.so -p 8085 -w /.../mjpg-streamer/www" -o "/.../mjpg-streamer/output_file.so -i /.../mjpg-streamer/input_uvc.so -m testvideo"`

The stream is viewable via the port specified, so that's awesome, but my `testvideo` is just a 10 second still image, not an actual video, from what I can tell. I've tried using both `input_uvc.so` and `/dev/video0` as the `-i` parameter, but neither works. What am I doing wrong??