jacksonliam/mjpg-streamer #187: support for YU12

jose1711, 2019-01-12:
I cannot get `mjpg-streamer` to work with Logitech QuickCam Zoom due to this:
```
# mjpg_streamer -i 'input_uvc.so -y -d /dev/video4'
MJPG Streamer Version: git rev: 43dc7a1413295de5b22868a2f49101064b2c5475
 i: Using V4L2 device.: /dev/video4
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: YUYV , driver gave us: YU12
    ... will try to handle this by checking against supported formats. 
 Init v4L2 failed !! exit fatal 
 i: init_VideoIn failed
```
Any idea?

jacksonliam, 2019-01-13:
What does `v4l2-ctl --list-formats` show?

jose1711, 2019-01-13:
```
$ v4l2-ctl --list-formats -d /dev/video4
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'PWC2' (Raw Philips Webcam Type (New), compressed)
	[1]: 'YU12' (Planar YUV 4:2:0)
```

jose1711, 2019-01-13:
[this fork](https://github.com/pranjalv123/mjpg-streamer-yu12) makes it possible to use this camera but having this feature merged into maintained mjpg-streamer codebase probably makes more sense.
output from a simple test:
```
./mjpg_streamer -i './input_uvc.so -d /dev/video4 -y -r QVGA' -o './output_http.so -p 1234'
MJPG Streamer Version: svn rev: Unversioned directory
 i: Using V4L2 device.: /dev/video4
 i: Desired Resolution: 320 x 240
 i: Frames Per Second.: 5
 i: Format............: YUV
 i: JPEG Quality......: 80
Adding control for Pan (relative)
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Tilt (relative)
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Pan Reset
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Tilt Reset
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Pan/tilt Reset
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Focus (absolute)
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
mapping control for Pan (relative)
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Tilt (relative)
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Pan Reset
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Tilt Reset
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Pan/tilt Reset
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Focus (absolute)
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for LED1 Mode
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for LED1 Frequency
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Disable video processing
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Raw bits per pixel
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
 o: www-folder-path...: disabled
 o: HTTP TCP port.....: 1234
 o: username:password.: disabled
 o: commands..........: enabled
GET /?action=stream HTTP/1.1
```

ooooona, 2020-08-06:
hi, I met the same problem, have you ever solved this problem?
Does it because mjpeg-streamer not support YU12(YUV420P), what can I do?

jose1711, 2020-08-06:
Nah, I get rid of the camera.

jquievreux, 2022-08-01:
@jose1711 this should work with => https://github.com/fcharron/mjpg-streamer