jacksonliam/mjpg-streamer #189: GPL license incompatibility

adrianschroeter, 2019-01-15:
It seems the source GPL licenses are incompatibel, for example mjpg_streamer.c is GPL v2 only (no "or later" clause) while plugins/input_ptp2/input_ptp2.c is GPLv3 or later.

This is blocking me to submit this as official package to the openSUSE distribution atm. 

Any way to get the permission from the authors to link the code?

Torxgewinde, 2019-01-15:
Hello Adrian,
I am the initial author of MJPG-streamer and I am OK with linking the code. I also send you an email. If there is anything required in addition to just making a statement here, please let me know.

Kind regards,
Tom Stöveken

jacksonliam, 2019-01-15:
In regards to changing the whole base to gplv3 I don't even know how to go about identifying potential owners let alone what the permission would look like to get it to gplv3... 

It might be possible to find who wrote the ptp2 code and get them to agree to gplv2. Is that the only one? I've fired off some messages to the two owners I can identify anyway.

The raspberry pi headers have their own permissive license but if you're not on a raspi it won't build in anyway.

Or you could build without the offending plugins. I doubt the ptp2 plugin is very useful anyways and you'd have to make gphoto2 a dependency of your package to include it.

adrianschroeter, 2019-01-15:
On Dienstag, 15. Januar 2019, 20:11:05 CET jacksonliam wrote:
> In regards to changing the whole base to gplv3 I don't even know how to go
> about identifying potential owners let alone what the permission would look
> like to get it to gplv3...

It would be enough to add the "or later" claim to the current GPLv2 source 
files. Atm they are GPLv2 only.

> It might be possible to find who wrote the ptp2 code and get them to agree
> to gplv2. Is that the only one? 

It seems to me (and our legal team).

> I've fired off some messages to the two
> owners I can identify anyway.

Yep, might be the easiest solution.

> The raspberry pi headers have their own permissive license but if you're not
> on a raspi it won't build in anyway.

IIRC that was a BSD style license which is GPLv2 compatible. GPLv3 is 
unfortunatly not v2 compatible.

> Or you could build without the offending plugins. I doubt the ptp2 plugin is
> very useful anyways and you'd have to make gphoto2 a dependency of your
> package to include it.

yes, that would be a solution (IIRC I would need also to remove the source 
file from our shipped source).

thanks a lot for the attention!
adrian

-- 

Adrian Schroeter
SUSE Linux Products GmbH,  Maxfeldstr. 5, 90409 Nuernberg, Germany
email: adrian@suse.de