jacksonliam/mjpg-streamer #190: 404 not found invalid input plugin number

LuckyLuzz, 2019-01-18:
Hey, I run into a problem with autostart script. Hope you could maybe help.
I installed the mjpg-streamer following commands:

```
sudo apt -y install libjpeg62-turbo-dev imagemagick libav-tools libv4l-dev cmake
sudo -iu pi
git clone https://github.com/jacksonliam/mjpg-streamer.git
cd mjpg-streamer/mjpg-streamer-experimental
export LD_LIBRARY_PATH=.
make
cd ~/mjpg-streamer/mjpg-streamer-experimental
./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"
or with
/usr/local/bin/mjpg_streamer -i "input_uvc.so -r 640x480 -d /dev/video0 -f 15 -q 80" -o "output_http.so -p 8080 -w /usr/local/share/mjpg-streamer/www"
```

```
MJPG Streamer Version: git rev: ddb69b7b4f114f3c2ca01adf55712792ca8aed43
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 15
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: JPEG
    ... will try to handle this by checking against supported formats.
    ... Falling back to the faster MJPG mode (consider changing cmd line options).
Setting FPS on the capture device is not supported, fallback to software framedropping
: Invalid argument
 i: Frame period time ......: 66 ms
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: /usr/local/share/mjpg-streamer/www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
```

till here everything is fine, but when I try to use a script to start the cam I get the "invalid input plugin number" message in firefox.

Script is:
`sudo nano /etc/init.d/mjpg_streamer`
```
-------------------------------------------------
#!/bin/sh
# /etc/init.d/mjpg_streamer
#
# Creation:    04.02.2013
# Last Update: 04.02.2013
#
# Written by Georg Kainzbauer (http://www.gtkdb.de)
#
### BEGIN INIT INFO
# Provides:          mjpg_streamer
# Required-Start:    $all
# Required-Stop:     $all
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: MJPG-Streamer
# Description:       MJPG-Streamer takes JPGs from Linux-UVC compatible webcams and streams them as M-JPEG via HTTP.
### END INIT INFO

start()
{
  echo "Starting mjpg-streamer..."
 LD_LIBARY_PATH=  /usr/local/bin/mjpg_streamer -i "/usr/local/lib/input_uvc.so -d /dev/video0 -n -y -r 640x480 -f 10" -o "/usr/local/lib/output_http.so -n -w /usr/local/www -p 8080" >/dev/null 2>&1 &
}
stop()
{
  echo "Stopping mjpg-streamer..."
  kill -9 $(pidof mjpg_streamer) >/dev/null 2>&1
}
case "$1" in
  start)
    start
    ;;
  stop)
    stop
    ;;
  restart)
    stop
    start
    ;;
  *)
    echo "Usage: $0 {start|stop|restart}"
    ;;
esac
exit 0
--------------------------------------------------
```
Commandline Message:

```
pi@orangepizero:~$ mjpg_streamer start

MJPG Streamer Version: git rev: ddb69b7b4f114f3c2ca01adf55712792ca8aed43
 o: www-folder-path......: disabled
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled

```

So the point is that I am not able to start the stream from a script.

OrangiPi H2
ARMBIAN 5.69 stable Debian GNU/Linux 9 (stretch) 4.19.13-sunxi

Any thoughts? 

Thanks! :)

http://www.gtkdb.de/index_36_2098.html
https://sebastien.andrivet.com/fr/posts/installing-multiple-octoprints-raspberry-pi-3-running-raspbian/

jacksonliam, 2019-01-18:
Hey, your web dirs are different.
e.g. -w /usr/local/www

Try changing the one in the script.

LuckyLuzz, 2019-01-18:
yeah there was som kind off dobble name thing. The script name was mjpg_streamer and the accual installed programm ist also called that name. So as I was thinking to start the script it started directly the streaming programm like I would typ:  /usr/local/bin/mjpg_streamer

ended up using crontab to call the script + gave the script a different name. 

sudo su
crontab -e
@reboot /etc/init.d/webcam_stream

strange but can be closed. thanks for your help!

SA7BNT, 2021-09-03:
Hey, im running in at 
404 not found invalid input plugin number
if i try to open "http://192.168.1.110:8080/?action=stream"
First install was working for me, but after a reboot i couldn't get it back up running. 

What may my issue be, something you can help me with?? 
The install steps are like yours.
 

output from mjpg-streamer.service

> ● mjpg-streamer.service - A Linux-UVC streaming application with Pan/Tilt
     Loaded: loaded (/etc/systemd/system/mjpg-streamer.service; disabled; vendor preset: enabled)
     Active: active (running) since Fri 2021-09-03 12:42:45 UTC; 4min 22s ago
   Main PID: 9911 (mjpg_streamer)
      Tasks: 2 (limit: 4182)
     Memory: 336.0K
     CGroup: /system.slice/mjpg-streamer.service
             └─9911 /usr/local/bin/mjpg_streamer

>Sep 03 12:42:45 localhost mjpg_streamer[9911]:  o: HTTP TCP port........: 8080
>Sep 03 12:42:45 localhost mjpg_streamer[9911]:  o: HTTP Listen Address..: (null)
>Sep 03 12:42:45 localhost mjpg_streamer[9911]:  o: username:password....: disabled
>Sep 03 12:42:45 localhost mjpg_streamer[9911]:  o: commands.............: enabled
>Sep 03 12:42:45 localhost mjpg_streamer[9911]: MJPG-streamer [9911]: www-folder-path......: disabled
>Sep 03 12:42:45 localhost mjpg_streamer[9911]: MJPG-streamer [9911]: HTTP TCP port........: 8080
>Sep 03 12:42:45 localhost mjpg_streamer[9911]: MJPG-streamer [9911]: HTTP Listen Address..: (null)
>Sep 03 12:42:45 localhost mjpg_streamer[9911]: MJPG-streamer [9911]: username:password....: disabled
>Sep 03 12:42:45 localhost mjpg_streamer[9911]: MJPG-streamer [9911]: commands.............: enabled
>Sep 03 12:42:45 localhost mjpg_streamer[9911]: MJPG-streamer [9911]: starting output plugin: output_http.so (ID: 00)

cd mjpg-streamer-experimental
make
sudo make install

then

sudo adduser webcam video
sudo usermod -a -G video webcam
init
mjpg-streamer.default   => /etc/default/mjpg-streamer
mjpg-streamer.init      => /etc/init.d/mjpg-streamer
After copying the above files run:

sudo update-rc.d mjpg-streamer defaults
sudo systemctl daemon-reload
systemd
mjpg-streamer.service   => /etc/systemd/system/mjpg-streamer.service

CountryDevil, 2022-01-18:
I am also getting the same thing a described above with the 404 error when trying to run as a service. 

`mjpg-streamer.service - A Linux-UVC streaming application with Pan/Tilt
   Loaded: loaded (/etc/systemd/system/mjpg-streamer.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2022-01-17 22:43:15 CST; 7s ago
 Main PID: 1783 (mjpg_streamer)
    Tasks: 2 (limit: 1935)
   CGroup: /system.slice/mjpg-streamer.service
           └─1783 /usr/local/bin/mjpg_streamer

Jan 17 22:43:15 PiDevil mjpg_streamer[1783]:  DBG(/home/pi/projects/mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer.c, main(), 411): starting 1 output plugin(s)
Jan 17 22:43:15 PiDevil mjpg_streamer[1783]:  DBG(/home/pi/projects/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/output_http.c, output_run(), 225): launching server thread #00
Jan 17 22:43:15 PiDevil mjpg_streamer[1783]: MJPG-streamer [1783]: MJPG Streamer Version: git rev: 310b29f4a94c46652b20c4b7b6e5cf24e532af39
Jan 17 22:43:15 PiDevil mjpg_streamer[1783]:  DBG(/home/pi/projects/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/httpd.c, server_thread(), 1555): waiting for clients to connect
Jan 17 22:43:15 PiDevil mjpg_streamer[1783]: MJPG-streamer [1783]: www-folder-path......: disabled
Jan 17 22:43:15 PiDevil mjpg_streamer[1783]: MJPG-streamer [1783]: HTTP TCP port........: 8080
Jan 17 22:43:15 PiDevil mjpg_streamer[1783]: MJPG-streamer [1783]: HTTP Listen Address..: (null)
Jan 17 22:43:15 PiDevil mjpg_streamer[1783]: MJPG-streamer [1783]: username:password....: disabled
Jan 17 22:43:15 PiDevil mjpg_streamer[1783]: MJPG-streamer [1783]: commands.............: enabled
Jan 17 22:43:15 PiDevil mjpg_streamer[1783]: MJPG-streamer [1783]: starting output plugin: output_http.so (ID: 00)
`