jacksonliam/mjpg-streamer #199: How to fix output resolution after I'm applying -roi

radu-at, 2019-02-28:
I'm starting the mjpeg_streamer with **-x 640 -y 480 -roi 0.0,0.4,1.0,1.0**, If I'm wright, the output image should now be recalculated based on ROI and must have on this case 640x288px on the output (initial height - 40% from ROI).

How can be fixed, because the player render it with the initial resolution of 640x480.

jacksonliam, 2019-02-28:
I believe ROI crops the sensor not the output image. If you want the output image to be a different resolution then set that as the x and y arguements.

closed 2019-02-28 by jacksonliam

radu-at, 2019-02-28:
On RaspberyPi -x & -y sets the sensor, I'm using  -x 640 -y 480 for full sensor size and the roi will crop from this image, if I'm using -x 640 -y 288 I will have the image with height 288px but from the middle of the sensor and not from the top-left/bottom-right

jacksonliam, 2019-02-28:
Have you tried it?

What about with -usestills ?

x and y sets the video width and height and also the crop width and height we send to the raspberry pi MMAL library. So that might be getting in your way.

If that is the case then I guess using ROI in this way is not supported.

You might be able to set the raspicam v4l2 driver to do what you want with v4l2-ctl and then use input_uvc to stream it with mjpg-streamer.