jacksonliam/mjpg-streamer #202: Kernel Panic

brunocerk, 2019-04-08:
Raspberry Pi 3 B+
Linux alarm 5.0.7-1-ARCH #1 SMP Sat Apr 6 00:09:17 UTC 2019 aarch64 GNU/Linux

processor       : **0 to 3**
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

> Installed mjpeg_streamer using git repo and the tutorial there.

i'm running this command : `mjpg_streamer -i "input_uvc.so -d /dev/video0 -f 15 -r 640x480" -o "output_http.so -w ./www"`

Output: 
MJPG Streamer Version: git rev: ddb69b7b4f114f3c2ca01adf55712792ca8aed43
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 15
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT

> Sometimes it brakes here

UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)

> But never pass this point

I was running the commands on SSH, so after a lot of crashes and linux installations i plugged a HDMI on raspberry to see if i was missing something.

And i came across this output on screen plugged to hdmi :
[image]

After several seconds other lines of error appear and thats what all i could get on my screen, the end of those lines 
[image]

Sorry for bad quality photo, i could not manage to find logs, i've tried `journalctl -b -1` to get last boot logs but all this lines don't appear there.

I have no clue of what it could be, i'm switching to Raspbian and i'll update here

yelvert, 2019-05-05:
Im getting this same bug as well, only found the kernel panic logs after plugging in a monitor.

yelvert, 2019-05-05:
Update: Switching back to 2018-11-13-raspbian-stretch-lite fixed the issue for me. It is failing on 2019-04-08-raspbian-stretch-lite.

jacksonliam, 2019-05-05:
Thanks both for the info I'll try and reproduce when I get a chance.