jacksonliam/mjpg-streamer #203: Make won't make input_raspicam.so plugin

Strat00s, 2019-04-18:
Title says it all. No matter what, it will never create input_raspicam.so plugin, so I can't use raspicam.
```
Scanning dependencies of target mjpg_streamer
[  4%] Building C object CMakeFiles/mjpg_streamer.dir/mjpg_streamer.c.o
[  8%] Building C object CMakeFiles/mjpg_streamer.dir/utils.c.o
[ 13%] Linking C executable mjpg_streamer
[ 13%] Built target mjpg_streamer
Scanning dependencies of target input_file
[ 17%] Building C object plugins/input_file/CMakeFiles/input_file.dir/input_file.c.o
[ 21%] Linking C shared library input_file.so
[ 21%] Built target input_file
Scanning dependencies of target input_http
[ 26%] Building C object plugins/input_http/CMakeFiles/input_http.dir/input_http.c.o
[ 30%] Building C object plugins/input_http/CMakeFiles/input_http.dir/misc.c.o
[ 34%] Building C object plugins/input_http/CMakeFiles/input_http.dir/mjpg-proxy.c.o
[ 39%] Linking C shared library input_http.so
[ 39%] Built target input_http
Scanning dependencies of target input_uvc
[ 43%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/dynctrl.c.o
[ 47%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/input_uvc.c.o
[ 52%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/jpeg_utils.c.o
[ 56%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/v4l2uvc.c.o
[ 60%] Linking C shared library input_uvc.so
[ 60%] Built target input_uvc
Scanning dependencies of target output_file
[ 65%] Building C object plugins/output_file/CMakeFiles/output_file.dir/output_file.c.o
[ 69%] Linking C shared library output_file.so
[ 69%] Built target output_file
Scanning dependencies of target output_http
[ 73%] Building C object plugins/output_http/CMakeFiles/output_http.dir/httpd.c.o
[ 78%] Building C object plugins/output_http/CMakeFiles/output_http.dir/output_http.c.o
[ 82%] Linking C shared library output_http.so
[ 82%] Built target output_http
Scanning dependencies of target output_rtsp
[ 86%] Building C object plugins/output_rtsp/CMakeFiles/output_rtsp.dir/output_rtsp.c.o
[ 91%] Linking C shared library output_rtsp.so
[ 91%] Built target output_rtsp
Scanning dependencies of target output_udp
[ 95%] Building C object plugins/output_udp/CMakeFiles/output_udp.dir/output_udp.c.o
[100%] Linking C shared library output_udp.so
[100%] Built target output_udp
```

It always says this when I try to run it.
```
MJPG Streamer Version: git rev: ddb69b7b4f114f3c2ca01adf55712792ca8aed43
ERROR: could not find input plugin
       Perhaps you want to adjust the search path with:
       # export LD_LIBRARY_PATH=/path/to/plugin/folder
       dlopen: input_raspicam.so: cannot open shared object file: No such file or directory
```
But there is no input_raspicam.so.

jacksonliam, 2019-04-18:
I assume you are building on a raspberry pi? What OS are you running?

Can you post the output of a full build (run make distclean first).

Strat00s, 2019-04-18:
> I assume you are building on a raspberry pi? What OS are you running?
> 
> Can you post the output of a full build (run make distclean first).

Can't now.
I am running Raspberry pi 3 A+ and DietPi.

jacksonliam, 2019-04-18:
If you are running a cut down OS it's possible it doesn't have the required videocore headers and libraries included. Look for the directories 
/opt/vc/lib and /opt/vc/include

Strat00s, 2019-04-18:
Here is full make:
```
root@DietPi:/home/dietpi/mjpg-streamer/mjpg-streamer-experimental# make
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMAKE_BUILD_TYPE = Release
-- Looking for include file sys/inotify.h
-- Looking for include file sys/inotify.h - found
CMake Warning at plugins/input_opencv/CMakeLists.txt:4 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
-- Could NOT find GPHOTO2 (missing:  GPHOTO2_LIBRARY GPHOTO2_INCLUDE_DIR) 
-- Looking for include file linux/videodev2.h
-- Looking for include file linux/videodev2.h - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Could NOT find SDL (missing:  SDL_LIBRARY SDL_INCLUDE_DIR) 
-- Could NOT find protobuf-c (missing:  PROTOBUF_C_LIBRARY PROTOBUF_C_INCLUDE_DIR) 
-- 
-- The following features have been enabled:

 * PLUGIN_INPUT_FILE , File input plugin
 * PLUGIN_INPUT_HTTP , HTTP input proxy plugin
 * PLUGIN_INPUT_UVC , Video 4 Linux input plugin
 * PLUGIN_OUTPUT_FILE , File output plugin
 * PLUGIN_OUTPUT_HTTP , HTTP server output plugin
 * PLUGIN_OUTPUT_RTSP , RTSP output plugin
 * PLUGIN_OUTPUT_UDP , UDP output stream plugin

-- The following OPTIONAL packages have been found:

 * Threads

-- The following features have been disabled:

 * WXP_COMPAT , Enable compatibility with WebcamXP
 * PLUGIN_INPUT_OPENCV , OpenCV input plugin (unmet dependencies)
 * PLUGIN_INPUT_RASPICAM , Raspberry Pi input camera plugin (unmet dependencies)
 * PLUGIN_INPUT_PTP2 , PTP2 input plugin (unmet dependencies)
 * ENABLE_HTTP_MANAGEMENT , Enable experimental HTTP management option
 * PLUGIN_OUTPUT_VIEWER , SDL output viewer plugin (unmet dependencies)
 * PLUGIN_OUTPUT_ZMQSERVER , ZMQ Server output plugin (unmet dependencies)
[... 180 lines cut ...]
-- Installing: /usr/local/share/mjpg-streamer/www/favicon.png
-- Installing: /usr/local/share/mjpg-streamer/www/style.css
-- Installing: /usr/local/share/mjpg-streamer/www/cambozola.jar
-- Installing: /usr/local/lib/mjpg-streamer/input_file.so
-- Installing: /usr/local/lib/mjpg-streamer/input_http.so
-- Installing: /usr/local/lib/mjpg-streamer/input_uvc.so
-- Installing: /usr/local/lib/mjpg-streamer/output_file.so
-- Installing: /usr/local/lib/mjpg-streamer/output_http.so
-- Installing: /usr/local/lib/mjpg-streamer/output_rtsp.so
-- Installing: /usr/local/lib/mjpg-streamer/output_udp.so
make[1]: Leaving directory '/home/dietpi/mjpg-streamer/mjpg-streamer-experimental/_build'
```

> If you are running a cut down OS it's possible it doesn't have the required videocore headers and libraries included. Look for the directories
> /opt/vc/lib and /opt/vc/include

I have no idea, what is supposed to be in those 2 directories, so here is all I can find in them:
```
root@DietPi:~# cd /opt/vc/lib
root@DietPi:/opt/vc/lib# ls
libbcm_host.so  libbrcmGLESv2.so  libbrcmWFC.so     libdebug_sym.so  libelftoolchain.so     libmmal_core.so  libmmal_util.so       libopenmaxil.so  libvcos.so  plugins
libbrcmEGL.so   libbrcmOpenVG.so  libcontainers.so  libdtovl.so      libmmal_components.so  libmmal.so       libmmal_vc_client.so  libvchiq_arm.so  libvcsm.so
root@DietPi:/opt/vc/lib# cd /opt/vc/include
-bash: cd: /opt/vc/include: No such file or directory
```

jacksonliam, 2019-04-18:
Yes you are missing /opt/vc/include. I'd suggest using a distro which includes it (e.g. raspbian) or asking your distro maintainers how they suggest you add it. Perhaps they have a package.

closed 2019-04-18 by jacksonliam

Strat00s, 2019-04-18:
Apparently, all I needed to do was to install libraspberrypi-dev package. Although I had to google for it.

naikymen, 2021-11-05:
I came across this issue by blindly copying commands from a tutorial, not realizing that the quotation marks around `input_raspicam.so` were actually some very similar symbol:

```bash
# Incorrect
mjpg_streamer -i “./input_raspicam.so” -o "output_http.so"
 
# Correct
mjpg_streamer -i "input_raspicam.so" -o "output_http.so"
```
Hope it may help someone as distracted as I :P

Thanks for the software!