jacksonliam/mjpg-streamer #207: ERROR: could not find input plugin

huazai665, 2019-05-04:
ERROR: could not find input plugin
       Perhaps you want to adjust the search path with:
       # export LD_LIBRARY_PATH=/path/to/plugin/folder
       dlopen: ./input_raspicam.so: cannot open shared object file: No such file or directory

jacksonliam, 2019-05-04:
What commands did you run?

huazai665, 2019-05-04:
./mjpg_streamer -i "./input_raspicam.so" -o "./output_http.so -w ./www"

jacksonliam, 2019-05-04:
And did you first run
`export LD_LIBRARY_PATH=.`
?

huazai665, 2019-05-04:
[image]
yes

jacksonliam, 2019-05-04:
It's probably not getting built, I assume you're building on a raspberry pi? Which OS did you build it on? 

When you build in the console output under the section 
`-- The following features have been disabled:`

Does it list 
 `* PLUGIN_INPUT_RASPICAM , Raspberry Pi input camera plugin (unmet dependencies)`
?

huazai665, 2019-05-04:
I have built it on raspberry 3B+ successfully and the OS is ubuntu mate 18.02.
[image]

jacksonliam, 2019-08-07:
Did you figure this out? I just noticed the argument should be "input_raspicam.so"  not "./input_raspicam.so".

closed 2019-08-07 by jacksonliam