jacksonliam/mjpg-streamer #215: What is the camera mode when streaming video on Raspberry Pi?

rolandjitsu, 2019-08-05:
I cannot seem to find what the is the camera [mode](https://www.raspberrypi.org/documentation/raspbian/applications/camera.md) set by the raspicam plugin when streaming a video.

Is it using the default mode (1920x1080) and then crops?

jacksonliam, 2019-08-05:
I believe it is auto, e.g we just pass the user supplied resolution and framerate to the MMAL API and the driver picks the correct 'mode'.

closed 2019-08-05 by jacksonliam

kodonnell, 2019-10-18:
We might be interested in forcing a mode - as a note to self, or anyone else interest, [this](https://github.com/raspberrypi/userland/blob/3617b3f5c096eec6ab9121a67dc1c28ef89fe8f9/host_applications/linux/apps/raspicam/RaspiStill.c#L817-L823) appears to be what's needed.