jacksonliam/mjpg-streamer #216: mjpg-streamer as client. Streaming to server with static ip.

UnderOverCovered, 2019-08-06:
Hello everyone, I want to make video stream from raspberry pi to PC with static IP via CSI camera and mjpg-streamer. I don't now how to do it without custom UDP forks. But custom UDP forks realization is really bad - I am receiving a damaged data. Also I don't wont to use custom TCP forks they are very slow.
So, can you help me please? Are there special mjpg-streamer's plugins?
So let me explain again in more details. Raspberry is a client, it hasn't static ip, there is no local network between raspi and PC. PC has a static IP, this PC have to receive a raspberry's video stream.
Thank you in advance for your responses

UnderOverCovered, 2019-08-06:
What about output_zmqserver? Why I can't run it? I am getting an error:
`Perhaps you want to adjust the search path with:
       # export LD_LIBRARY_PATH=/path/to/plugin/folder
       dlopen: output_zmqserver.so: cannot open shared object file: No such file or directory
`
I tried to do this:
`export LD_LIBRARY_PATH=/usr/src/mjpg-streamer/mjpg-streamer-experimental/plugins/output_zmqserver
`
Also I installed this packages: libzmq3-dev,  libprotobuf-c-dev libprotobuf-c1 libprotoc10 protobuf-c-compiler

jacksonliam, 2019-08-07:
You want to set the LD library path to the location of the .so files. This is usually the mjpg-streamer-experimental directory.

Do you have an output_zmqserver.so file anywhere?

If you installed the dependencies after running make the first time then do a `make distclean` to clear the cmake cache and try building again.

If it still doesn't work after that, post the full output of the build.

closed 2019-08-07 by jacksonliam

UnderOverCovered, 2019-08-07:
Hello! Thank you very much Jacksonliam! Yeah, I hadn't the output_zmqserver.so file in experimental directory. So I deleted the mjpg-streamer. Then after git download, I ran:
`cd mjpg-streamer-experimental`
`make distclean`
`make`
`sudo make install`
And yes this .so file in the folder. So, I ran mjpg-streamer with output_zmqserver - h and that works.
Thank you very much! Your streamer is the best!