jacksonliam/mjpg-streamer #217: Couldn't create zmq socket

UnderOverCovered, 2019-08-09:
Hello again, can you help me please with this issue? I  am starting the stream with zmq output and raspicam input.
`./mjpg_streamer -o "output_zmqserver.so --address 192.168.1.200:10000" -i "input_raspicam.so -x 640 -y 480 -fps 20 -ex night"
`
But after that I got the error `Couldn't create zmq socket`
I am using libzmq3-dev, because a couldn't download libzmq4-dev, that you recommend. Is this a zmq lib error? If it is, can you show me please the repo, where I can download this package?
I also tried to `make distclean`, removing libzmq3-dev , installing libzmq-dev and libzmq1, but after that I couldn't to `make` correctly. I can show you this big error log, if it necessary.
Thank you in advance.

UnderOverCovered, 2019-08-09:
Also, I searched this link [,](https://machinekit.narkive.com/xqNkEmkU/updating-machinekit-on-beaglebone-configure-failure-unable-to-locate-package-libzmq4-dev) where perhaps I can get this libzmq4-dev. But it returns 404. Or is it not lib error?

jacksonliam, 2019-08-09:
Cannot help with getting the libzmq4 dependency on your platform, that's not an mjpg-streamer issue.

If it compiles with libzmq3 then it's _probably_ ok anyway.

I don't think the address you are providing is a valid zmq URL, I may be wrong though, I'm not that familiar with zmq.

UnderOverCovered, 2019-08-09:
Thank you, I will try something with server. Maybe it helps. But, can I ask you, what is the type of the socket and protocol, that mjpg-streamer using while streaming with output_zmq? Because zmq includes really too much types of the sockets and transport protocols. Please...

closed 2020-12-18 by jacksonliam