jacksonliam/mjpg-streamer #218: input_init() return value signals to exit

Komak57, 2019-08-22:
This error appears to happen a lot with mjpg-streamer, and with different cameras (Tested with c920 and Sonix 2.0 Camera).

For now, we'll discuss the specs and issues, and some of the remedies I've tried for the 0c45:62c0 Sonix 2.0 Camera, which refuses to work.

OS: Ubuntu 16.04.6 [4.4.0-142-generic]
CPU: AMD FX-4100 Quad Core
GPU: Radeon HD5670
CAM: 0c45:62c0 Microdia Sonix USB 2.0 Camera
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912

    v4l2-ctl --list-formats-ext
    ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUYV 4:2:2
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 1.000s (1.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 1.000s (1.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 1.000s (1.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 1.000s (1.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 1.000s (1.000 fps)
                Size: Discrete 1280x960
                        Interval: Discrete 0.111s (9.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 1.000s (1.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.111s (9.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                        Interval: Discrete 1.000s (1.000 fps)

`mjpg_streamer -i "input_uvc.so -d /dev/video0 -r 1280x1024 -f 9" -o "output_http.so -p 8080 -w /var/www/mjpg-www"`

    i: Using V4L2 device.: /dev/video0
    i: Desired Resolution: 1280 x 1024
    i: Frames Per Second.: 9
    i: Format............: JPEG
    i: TV-Norm...........: DEFAULT
    i: Could not obtain the requested pixelformat: MJPG , driver gave us: YUYV
        ... will try to handle this by checking against supported formats.
        ... Falling back to YUV mode (consider using -yuv option). Note that this requires much more CPU power
[... 67 lines cut ...]
I've also copied the output_http.so and input_uvs.so to /urs/local/lib and set this directory to the LD_LIBRARY_PATH to no avail.

So let's check FFMPEG and see what I can see from there?

    ffmpeg -f video4linux2 -list_formats all -i /dev/video0

    ffmpeg version 2.8.15-0ubuntu0.16.04.1 Copyright (c) 2000-2018 the FFmpeg developers
    built with gcc 5.4.0 (Ubuntu 5.4.0-6ubuntu1~16.04.10) 20160609
    configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
    libavutil      54. 31.100 / 54. 31.100
    libavcodec     56. 60.100 / 56. 60.100
    libavformat    56. 40.101 / 56. 40.101
    libavdevice    56.  4.100 / 56.  4.100
    libavfilter     5. 40.101 /  5. 40.101
    libavresample   2.  1.  0 /  2.  1.  0
    libswscale      3.  1.101 /  3.  1.101
    libswresample   1.  2.101 /  1.  2.101
    libpostproc    53.  3.100 / 53.  3.100
    [video4linux2,v4l2 @ 0x240b360] Raw       :     yuyv422 :           YUYV 4:2:2 : 640x480 352x288 320x240 176x144 160x120 1280x960 1280x1024

At this point, I'm left with the conclusion that something is horribly wrong with my mjpg-streamer compilation.

Some additional information, using the Logitech c920, I can get video without a segment fault, but then I run into problems trying to get good quality from the camera at 1080p, and using -y or -q repeats the input_init() bug.

I have a similar setup using a slightly newer USB 2.0 camera on a raspberry pi (motioneye), and I have a bigger problem with USB bus speeds than compatibility issues, but it uses a different branch of mjpg-streamer for ARM devices.

jacksonliam, 2019-08-22:
Thanks for the thorough report, I'm assuming this is a compatibility issue between mjpg-streamer and Ubuntu's version of v4l2. It looks like its a problem with the YUV capture mode only, since I guess when the C920 works it's working in JPEG mode.

Are you able to get a valid capture (e.g you can watch it) from the Sonix 2.0 Camera in another app, e.g. ffmpeg?

jacksonliam, 2019-08-22:
Actually, scratch that.

The input_init message is because it doesn't recognize one of the options you've passed (error could be clearer there).

I expect you've built mjpg-streamer without libjpeg causing it to not support YUV mode (again, error could be better, we probably shouldn't try YUV mode automatically if we can't support it).

Can you make sure you have the 
libjpeg dependencies installed (see readme.md) and do a `make distclean` before building again.

Komak57, 2019-08-22:
Can confirm that FFMPEG can pull data from 0c45:62c0 Microdia Sonix USB 2.0 Camera.
Can confirm that libjpeg8-dev is the latest version (8c-2ubuntu8) before recompile. Wouldn't originally compile without it.

Note* Running the c920 AND the USB 2.0 Camera simultaneously appears to be impossible at the moment?

I'd also like to note that, in the compilation, only `input_file.so`, `input_http.so`, and `input_uvc.so` appear to be built, while `plugins/input_opencv` DOES exist.

After a recompile with `make distclean` everything appears to be working. Including quality options for both the USB 2.0 and c920.

dstulken, 2019-08-25:
I can confirm having the same issue, with the same symptoms, and also the same result of a 'make distclean' plus a rebuild producing a working executable.

This was on a LMDE 3 Cindy system (Linux Mint Debian Edition).  

In my case, the error was quite confusing, as all of my cameras supported both MJPG and YUV modes.  So they would operate just fine in the "normal" case, it was only when I added the '-y' parameter to switch them to YUV mode that I received the cryptic input_init() failure response.  So it really did appear to be a command line option/parameter issue, but I double-checked it about a hundred times of course before my web searches led me here.

Is the lack of YUV support something that could be detected, and a different error message provided to the user when this condition occurs?

jacksonliam, 2019-08-25:
Yep error message should be better.

I believe this happens because you build for the first time without the libjpg dependency installed. CMake caches that you don't have it so any further builds would be without it untill you do a `distclean`.

I want to make this obvious but not sure how, without running `make` forcing a distclean each time, which would be non standard. Perhaps I can cause make to output some useful info when building without libjpeg...

morganrallen, 2019-09-11:
I'd like to report I had this same issue on a RPi4 with Buster (2019-07-10).

During my first build it was reported libjpeg.h could not be found, I install `libjpeg-dev` and carried on with the build, resulting in the same errors as above. **To get it working**, after installing `libjpeg-dev`,  I deleted my build directory, reran `cmake`, rebuilt and it worked as expected.

Thanks for the tool, I looks really great with a (even low end) camcorder and HDMI>USB3 capture card.

DangerManM, 2024-11-13:
First, "make clean && make uninstall", then, "make && make install", and finally, remember to add the "-y" command to the start.sh