jacksonliam/mjpg-streamer #219: CV_IMWRITE_JPEG_QUALITY，this place error

gonepoo, 2019-08-26:
Scanning dependencies of target input_opencv
[ 31%] Building CXX object plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o
/usr/local/mjpg/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:86:5: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
     " Help for input plugin..: "INPUT_PLUGIN_NAME"\n" \
     ^
/usr/local/mjpg/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp: In function ‘void* worker_thread(void*)’:
/usr/local/mjpg/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:409:34: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
     compression_params.push_back(CV_IMWRITE_JPEG_QUALITY);
                                  ^~~~~~~~~~~~~~~~~~~~~~~
/usr/local/mjpg/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:409:34: note: suggested alternative: ‘IN_CMD_JPEG_QUALITY’
     compression_params.push_back(CV_IMWRITE_JPEG_QUALITY);
                                  ^~~~~~~~~~~~~~~~~~~~~~~
                                  IN_CMD_JPEG_QUALITY
make[2]: *** [plugins/input_opencv/CMakeFiles/input_opencv.dir/build.make:63：plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o] error 1
make[1]: *** [CMakeFiles/Makefile2:243：plugins/input_opencv/CMakeFiles/input_opencv.dir/all] error 2
make: *** [Makefile:130：all] error 2

jacksonliam, 2019-08-26:
What OS are you building on? What version of OpenCV do you have installed?

gonepoo, 2019-08-27:
Raspberry Pi  4 Buster，and I use the latest version of openCV， I clone from this link “https://github.com/opencv/opencv “
I also install the OpenCV_contrib from this link “https://github.com/opencv/opencv_contrib“

发送自 Windows 10 版邮件应用

发件人: jacksonliam
发送时间: 2019年8月26日 22:32
收件人: jacksonliam/mjpg-streamer
抄送: gonepoo; Author
主题: Re: [jacksonliam/mjpg-streamer] CV_IMWRITE_JPEG_QUALITY，this place error (#219)

What OS are you building on? What version of OpenCV do you have installed?

gonepoo, 2019-08-27:
I just want the base function for home security， thank you for the help， I will try it  

gonepoo@163.com
 
发件人： jacksonliam
发送时间： 2019-08-27 14:25
收件人： jacksonliam/mjpg-streamer
抄送： gonepoo; Author
主题： Re: [jacksonliam/mjpg-streamer] CV_IMWRITE_JPEG_QUALITY，this place error (#219)
That looks like OpenCV 4.x, it's not supported. You need 3.x as specified in the docs
https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_opencv/README.md
Do you want to use the OpenCV plugin? If not you can just remove add_subdirectory(plugins/input_opencv) from CMakeLists.txt and do a make distclean before make

jacksonliam, 2019-08-27:
That looks like OpenCV 4.x, it's not supported. You need 3.x as specified in the docs 
https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_opencv/README.md

Do you want to use the OpenCV plugin? If not you can just remove `add_subdirectory(plugins/input_opencv)` from CMakeLists.txt and do a `make distclean` before `make`

gonepoo, 2019-08-27:
Thank you very much , it really works

| |
高晶
邮箱：gonepoo@163.com
|

签名由 网易邮箱大师 定制

On 08/27/2019 17:46, jacksonliam wrote:

Closed #219.

closed 2019-08-27 by jacksonliam

gonepoo, 2019-08-27:
when I used the command “sudo make” ， it occurs this error

gonepoo, 2019-08-27:
Thanks a lot , it works