jacksonliam/mjpg-streamer #223: cropping before jpg conversion?

mark-hahn, 2019-10-06:
I using a raspi w with mjpg-streamer (just like octopi) and it works great. In my webpage showing the stream I often need to zoom in, i.e. crop the image. The problem with this is that I still have to stream the full-size image which is somewhat slow. It would be ideal if I could crop the image before it is converted to jpeg.

I know the jpeg processing happens inside the broadcom chip before any software gets the image. Can I crop the image inside the chip before it is converted to jpeg?

jacksonliam, 2019-10-07:
Just setting the stream width/height should crop.

Else perhaps the roi argument achieves what you're after `-roi 0.5,0.5,0.25,0.25` (see raspberry pi camera docs for how this works).

closed 2019-10-07 by jacksonliam

mark-hahn, 2019-10-09:
Thanks, ROI is just what I needed.