jacksonliam/mjpg-streamer #228: exec format error during docker build

siochs, 2019-11-06:
```bash
$ docker build -t mjpegstreamer .
Sending build context to Docker daemon   4.12MB
Step 1/9 : FROM chunliu/docker-opencv
latest: Pulling from chunliu/docker-opencv
b3e1c725a85f: Pull complete
...
Digest: sha256:05663cc75f47aec5ffc5738df966768b0204153ebd99a202aa7f2f191d415a12
Status: Downloaded newer image for chunliu/docker-opencv:latest
 ---> 6abcccc1d499
Step 2/9 : MAINTAINER Chun Liu <https://github.com/chunliu>
 ---> Running in fe5312783566
Removing intermediate container fe5312783566
 ---> 42babdc60f5d
Step 3/9 : LABEL Description="A Docker image for mjpg_streamer." Version="0.3"
 ---> Running in 1ce0beaacd8a
Removing intermediate container 1ce0beaacd8a
 ---> 2d5b4c9d9cf0
Step 4/9 : RUN git clone https://github.com/jacksonliam/mjpg-streamer.git
 ---> Running in 6e463f81def6
standard_init_linux.go:211: exec user process caused "exec format error"
The command '/bin/sh -c git clone https://github.com/jacksonliam/mjpg-streamer.git' returned a non-zero code: 1
```
I guess that the base image deviated from is not compatible.
Platform used: Raspbian Buster on Raspberry Pi 4b rev. 1.1

jacksonliam, 2020-02-29:
@chunliu any ideas?

chunliu, 2020-03-01:
From the error message, `git clone https://github.com/jacksonliam/mjpg-streamer.git` failed for some reason. I'd suggest you check if `git` is installed, and if it is installed, try to run the above `git clone` command in the shell and see if it would fail too.

VanDavv, 2020-03-27:
@siochs I have this error whenever I try to run e.x. 64 bit container on ARM or vice versa - maybe this will help you to find a clue what's going on

chunliu, 2020-03-28:
@VanDavv thanks for reminding me. The base image this container based on is derived from ubuntu:16.04 which might not be compatible with ARM. I never tested it with ARM anyway. You can find the source of the base image here: [https://github.com/chunliu/docker-opencv](https://github.com/chunliu/docker-opencv). Please feel free to fork it and make it compatible with ARM.

VanDavv, 2020-03-28:
For the container to be compatible on ARM, it needs to be compiled on ARM.

And the command that @siochs ran, `git clone`, gave an error identical to the one you receive when running e.x. ARM container in x86_64 environment. 

Example:
```
$ uname -m
x86_64
$ docker run --rm armhf/debian bash
standard_init_linux.go:211: exec user process caused "exec format error"
```

siochs, 2020-07-16:
Sowhere in read that docker build can now also something like "cross build" (for other platforms). But I cant find the reference anymore...

closed 2020-12-18 by jacksonliam