jacksonliam/mjpg-streamer #229: Unable to set format: 1196444237 res: 1920x1080

rclsilver, 2019-11-10:
Hi,

Local build of MJPG-Streamer from sources. Unable to view video.

Running on Raspberry Pi 4 with Raspbian GNU/Linux 10 (buster).

mjpg_streamer version:
```
mjpg-streamer@rpi-octoprint:~/mjpg-streamer/mjpg-streamer-experimental $ ./mjpg_streamer --version
MJPG Streamer Version: 501f6362c5afddcfb41055f97ae484252c85c912
mjpg-streamer@rpi-octoprint:~/mjpg-streamer/mjpg-streamer-experimental $ 
```

Camera formats list:
```
mjpg-streamer@rpi-octoprint:~/mjpg-streamer/mjpg-streamer-experimental $ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
	[1]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 1280x720
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.333s (3.000 fps)
mjpg-streamer@rpi-octoprint:~/mjpg-streamer/mjpg-streamer-experimental $ 
```

Full V4L details:
```
mjpg-streamer@rpi-octoprint:~/mjpg-streamer/mjpg-streamer-experimental $ v4l2-ctl --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : papalook FHD Camera: papalook F
	Bus info         : usb-0000:01:00.0-1.3
	Driver version   : 4.19.75
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : papalook FHD Camera: papalook F
	Serial           : 
	Bus info         : 1.3
	Media version    : 4.19.75
	Hardware revision: 0x00000100 (256)
	Driver version   : 4.19.75
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : papalook FHD Camera: papalook F
	Function         : V4L2 I/O
	Flags         : default
	Pad 0x01000007   : 0: Sink
	  Link 0x02000010: from remote pad 0x100000a of entity 'Extension 3': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1280/720
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 1843200
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1280, Height 720
	Default     : Left 0, Top 0, Width 1280, Height 720
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 1280, Height 720, Flags: 
Selection: crop_bounds, Left 0, Top 0, Width 1280, Height 720, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=0 value=0
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=30 value=30
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=100 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=300 step=1 default=100 value=100
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=1 max=7 step=1 default=1 value=1
         backlight_compensation 0x0098091c (int)    : min=0 max=3 step=1 default=2 value=2
mjpg-streamer@rpi-octoprint:~/mjpg-streamer/mjpg-streamer-experimental $ 
```

mjpg_streamer output:
```
mjpg-streamer@rpi-octoprint:~/mjpg-streamer/mjpg-streamer-experimental $ ./mjpg_streamer -i './input_uvc.so -r 1920x1080 -q 100 -d /dev/video1' -o './output_http.so -w ./www -p 8081 -l 127.0.0.1' 
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 1920 x 1080
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
Unable to set format: 1196444237 res: 1920x1080
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
mjpg-streamer@rpi-octoprint:~/mjpg-streamer/mjpg-streamer-experimental $ 
```

Same webcam works on Raspberry Pi 3, with raspbian 9 (stretch).

jacksonliam, 2020-02-29:
I wonder if the change in raspbian version has changed something in your cameras driver or perhaps v4l2 itself.

Have you tried different resolutions?

I assume you cannot run earlier raspbian versions on an rpi 4 as they don't have the hardware support?

closed 2020-12-18 by jacksonliam

3DStreetFurniture, 2022-10-25:
Unable to set format: 1196444237 res: 1280x720
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed

Need help with the above please. I'm using a logitech c920

alexfornuto, 2023-03-09:
FWIW I hit this issue because other software was already using the camera.

antoxaredst, 2023-11-11:
> Кстати, я столкнулся с этой проблемой, потому что другое программное обеспечение уже использовало камеру.

its real help me.. thanks))

alexfornuto, 2023-11-13:
> > Кстати, я столкнулся с этой проблемой, потому что другое программное обеспечение уже использовало камеру.
> 
> its real help me.. thanks))

Рад был помочь

Zotter, 2025-04-19:
I concur after upgraded to Fedora 42 - been using mjpg-steramer for more than a few versions prior - routine is to update my self compiled binaries, JIC. 

However, when I tried to run it from command line - got this error. Once I figured out it was running as I'd recently added a systemctl script, so used systemctl to stop - then restarted. All is good, wokrs as epected. Problem was a running instance while trying to start a new instance - not the new compile I'd just installed.

alexfornuto, 2025-04-21:
> FWIW: "Кстати, я столкнулся с этой проблемой, потому что другое программное обеспечение уже использовало камеру." translates to: "By the way, I encountered this problem because another software was already using the camera."

Yes, it's the translation (loosely) of my comment above it.