jacksonliam/mjpg-streamer #233: ov2710 error

DaTTcz, 2019-11-22:
Hello, i can't start the ov2710 camera. What am I doing wrong ? Thank you.

**lsusb**
`ID 0c45: 6368 Microdia`

**./mjpg_streamer -i "./input_uvc.so -r 1920x1080 -f 30" -o "./output_http.so"**
```
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using the V4L2 device: / dev / video0
 i: Desired Resolution: 1920x1080
 i: Frames Per Second
 i: Format ............: JPEG
 i: TV Norm ...........: DEFAULT
 i: FPS coerced ......: from 30 to 10000000
Unable to Map Buffer: No such device
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
```

**v4l2-ctl -list-formats-ext**
```
ioctl: VIDIOC_ENUM_FMT
        Index: 0
        Type: Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name: MJPEG
                Size: Discrete 1920x1080
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 432x240
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 320x184
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 160x120mm
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 960x720
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (31.000 fps)
                Size: 1280x960
                        Interval: Discrete 0.03s (30.000 fps)

        Index: 1
        Type: Video Capture
        Pixel Format: 'YUYV'
        Name: YUV 4: 2: 2
                Size: Discrete 1920x1080
                        Interval: Discrete 0.200s (5,000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 432x240
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 320x184
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 160x120mm
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.03s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.045s (21.000 fps)
                Size: Discrete 960x720
                        Interval: Discrete 0.093s (11.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.075s (13.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.113s (9.000 fps)
                Size: 1280x960
                        Interval: Discrete 0.167s (6.000 fps)
```

jacksonliam, 2020-02-29:
The `i: FPS coerced ......: from 30 to 10000000` part is very strange! Not sure that is an mjpg-streamer issue though.

Have you tried different resolutions, not specifying frame rate or yuv mode?

jasongoldsmith, 2020-03-02:
I just got the same error using a Logitech C920 - I have been messing around with U4VL shortly before this - I wonder if the other person above has that installed - I'm leaning towards that being the issue - but I don't know how to determine that for sure.

Starting up webcamDaemon...

--- Configuration: ----------------------------
camera:        auto
usb options:   -r 1920x1080 -f 30
raspi options: -fps 10
http options:  -w ./www-octopi -n
-----------------------------------------------

Found video devices:
/dev/video0
/dev/video1
/dev/video10
/dev/video11
/dev/video12
/dev/video20
/dev/video21
/dev/video22
USB device was not set in options, start MJPG-streamer with the first found video device: /dev/video0
<13>Mar  1 21:39:53 pi: Starting USB webcam
Running ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_uvc.so -r 1920x1080 -f 30 -d /dev/video0
MJPG Streamer Version: git rev: ddb69b7b4f114f3c2ca01adf55712792ca8aed43
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1920 x 1080
 i: Frames Per Second.: 30
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: FPS coerced ......: from 30 to 10000000
Unable to map buffer: No such device
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed

jacksonliam, 2020-03-02:
Possibly something else is grabbing the video device or perhaps mjpg-streamer doesnt have the permissions to access the device due to user/group or v4l config permission changes.

closed 2020-12-18 by jacksonliam