jacksonliam/mjpg-streamer #234: Orange Pi PC camera doesn't work after "make clean all"

kreston-ru, 2019-12-25:
Hello, dear friend!
I tried start video stream under your software, but it's doesn't work... :(

Please, can do you to say me: where is my mistake?

Board: Orange Pi PC Plus (Armbian installed).
Camera: OV5640 from Steven Zhao (AliExpress - Orange Pi official camera).

It's working under ffmpeg, but latency is about 3 second! But i make robot. And me need low latency. 3 second - it's very long time. I read, that your software can to give low latency of camera, but I cant to get it work... :(

after "make clean all", I have next files:
`$ ls
_build         
Dockerfile       
input_http.so  
makedeb.sh     
mjpg_streamer.c         
output_file.so  
output_udp.so   
README.md  
start.sh.bak  
utils.h
cmake           
docker-start.sh 
input_uvc.so   
Makefile       
mjpg_streamer.h         
output_http.so  
plugins         
scripts    
TODO          
www
CMakeLists.txt 
input_file.so    
LICENSE        
mjpg_streamer  
mjpg_streamer@.service  
output_rtsp.so  
postinstall.sh  
start.sh   
utils.c`

My " v4l2-ctl --list-formats"
`~/temp/ms/mjpg-streamer-master/mjpg-streamer-experimental$ v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: '422P'
        Name        : planar YUV 422

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'YU12'
        Name        : planar YUV 420

        Index       : 2
        Type        : Video Capture
        Pixel Format: 'YV12'
        Name        : planar YVU 420

        Index       : 3
        Type        : Video Capture
        Pixel Format: 'NV16'
        Name        : planar YUV 422 UV combined

        Index       : 4
        Type        : Video Capture
        Pixel Format: 'NV12'
        Name        : planar YUV 420 UV combined

[... 461 lines cut ...]
                Size: Discrete 640x480
                Size: Discrete 320x240
                Size: Discrete 176x144

        Index       : 23
        Type        : Video Capture
        Pixel Format: 'BA12'
        Name        : RAW Bayer RGGB 12bit
                Size: Discrete 2592x1936
                Size: Discrete 2048x1536
                Size: Discrete 1920x1080
                Size: Discrete 1600x1200
                Size: Discrete 1280x960
                Size: Discrete 1280x720
                Size: Discrete 1024x768
                Size: Discrete 800x600
                Size: Discrete 640x480
                Size: Discrete 320x240
                Size: Discrete 176x144

`
Can do you help me?

P.S.
Camera connected via CSI.

jacksonliam, 2019-12-26:
What command are you running and what is the error?

kreston-ru, 2019-12-28:
I tried to run via start.sh. I previously configured it...

1 test: ./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so -w ./www"

Next I changed it so:
`cd ~/temp/ms/mjpg-streamer-master/mjpg-streamer-experimental`
and next tests I tried to make with string above...

2 test: `./mjpg_streamer -o "./output_http.so -w ./www" -i "./input_raspicam.so -x 640 -y 480 -fps 10 -ex auto -awb auto -vs -ISO 100"`

3 test: `./mjpg_streamer -i "./input_uvc.so -n -f 30 -r 1280x960"  -o "./output_http.so -w ./www"`

4 test: `./mjpg_streamer -i "./input_uvc.so -n -f 30 -r 640x480 -d /dev/video0"  -o "./output_http.so -w ./www" &`

5 test: `./mjpg_streamer -i "./input_uvc.so -n -y -f 30 -r 640x480 -d /dev/video0"  -o "./output_http.so -w ./www" &`

6 test: `./mjpg_streamer -i "./input_uvc.so -n -y -f 30 -r 320x240 -d /dev/video0"  -o "./output_http.so -w ./www"`

I saw this error every time when try to run it!

jacksonliam, 2019-12-28:
What is the error?

kreston-ru, 2019-12-28:
`$ ./start.sh
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 320 x 240
 i: Frames Per Second.: 30
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
libv4l2: error setting pixformat: Operation not permitted
Unable to set format: 1448695129 res: 320x240
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
`

jacksonliam, 2019-12-28:
Try changing -y to -u (or just adding -u if -y not there)

kreston-ru, 2019-12-28:
It's started!!!!!
Thank you, friend! I'll be testing!

kreston-ru, 2019-12-28:
Friend! It's working! But how to flip image?

kreston-ru, 2019-12-28:
Thank you, friend! Delay is about 0.3sec!!! Your soft is perfect!

closed 2019-12-28 by jacksonliam

kreston-ru, 2019-12-28:
Dear @jacksonliam! Help me still one time, please! 
I tried to use -vf key, but it's not working!
I tried to use -rot 180, but it's not working too...
May be exist still one variant to flip image?

jacksonliam, 2019-12-28:
Probably not supported by your camera then.
Can try running something like `v4l2-ctl --set-ctrl vertical_flip=1` in the terminal before mjpg-streamer but probably won't work either.

If you know c you could try modifying the libjpeg code to flip the image.

Otherwise best bet is to get the destination to flip it when it's displayed.

kreston-ru, 2019-12-28:
Ok! Thank you very much!

kreston-ru, 2019-12-28:
still one question:
Included in soft http-server understand php? how can configure it ?

jacksonliam, 2019-12-28:
No php, I think there is a cgi script support but have not personally used it and not sure how secure it is.

kreston-ru, 2019-12-29:
Ok! I think better will be install apache with php and with MySql... It will be normal working together (parallel) with your web server?
P.S. 
Thank you still time for develop this software! :) It's amazing! I drive of my robot without delay of video :)

kreston-ru, 2019-12-29:
Still 1 question :) 
If I add -d parameter, it will register daemon in system?

kreston-ru, 2019-12-29:
I made `sudo make install`, but `systemctl` do not found mjpg_streamer....

jacksonliam, 2019-12-29:
No -d doesn't do that, 
Make install doesn't install the systemd stuff.
you have to install the systemd unit file first (after editing in your options).