jacksonliam/mjpg-streamer #236: Snapshot image is corrupt

schneemaier, 2020-01-08:
Hi,

i am trying to grab an image and then use convert add date and time to the image. I use 
`wget http://user:password@localhost:8081/?action=snapshot -O test.jpg`
to get the jpeg and then
`convert ./test.jpg -quality 90 -gravity SouthEast -pointsize 22 -fill white -annotate +0+0 "TEST" ./test2.jpg`
to add the text to the image. The problem is that convert outputs a bunch of warnings. It still converts the image but it happens much slower then normally.
Here are the warnings:
`onvert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd0 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd3 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd6 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd4 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 2 extraneous bytes before marker 0xd1 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 2 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd6 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 3 extraneous bytes before marker 0xd2 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd3 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 2 extraneous bytes before marker 0xd6 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 4 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 3 extraneous bytes before marker 0xd0 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd4 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd4 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 4 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 2 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 4 extraneous bytes before marker 0xd0 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd3 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd2 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd3 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd6 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd4 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 2 extraneous bytes before marker 0xd1 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 2 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd6 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 3 extraneous bytes before marker 0xd2 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd3 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 2 extraneous bytes before marker 0xd6 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 4 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 3 extraneous bytes before marker 0xd0 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd4 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd4 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 4 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 2 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 4 extraneous bytes before marker 0xd0 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd3 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd2 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd3 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd6 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd7 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd4 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 2 extraneous bytes before marker 0xd1 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 2 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: Corrupt JPEG data: 1 extraneous bytes before marker 0xd5 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
[... cut ...]
convert-im6.q16: Corrupt JPEG data: 3 extraneous bytes before marker 0xd2 `./test.jpg' @ warning/jpeg.c/JPEGWarningHandler/387.
convert-im6.q16: too many exceptions (exception processing is suspended) @ error/exception.c/ThrowException/974.`

Not sure where the issue is coming from...

I am running Buster 10.2 with kernel 4.19.75+. 

Webcam used: ID 046d:0823 Logitech, Inc. HD Webcam B910

Any idea how can i fix this problem?

Thanks
 

 convert ./test.jpg -quality 90 -gravity SouthEast -font /usr/share/fonts/truetype/ttf-dejavu/DejaVuSans.ttf -pointsize 22 -fill white -annotate +0+0 "TEST" ./test2.jpg

jacksonliam, 2020-01-08:
What command are you running mjpg-streamer with?

schneemaier, 2020-01-08:
This is the command:

/usr/local/bin/mjpg_streamer -b -i "/usr/local/lib/mjpg-streamer/input_uvc.so -d /dev/video2 -r 960x720 -f 10 -l auto" -o "/usr/local/lib/mjpg-streamer/output_http.so -p 8081 -w /usr/local/share/mjpg-streamer/www -c user:passwd"

I use the same command on other debian (not raspberry) arm installation and don't have the same problem.

jacksonliam, 2020-01-09:
In this mode we are taking jpegs straight from the camera and not touching them so it's most likely either the camera or it's driver causing jpeg weirdness.

You could try yuv mode (adding -y to the input plugin) as then we create the jpegs in mjpg-streamer, obviously this increases the CPU usage of mjpg-streamer.

closed 2020-01-09 by jacksonliam

dwery, 2021-03-14:
I've hit the same with a logitech C270. Would it be possible to remove the extra data within mjpg-streamer?

jacksonliam, 2021-03-15:
> I've hit the same with a logitech C270. Would it be possible to remove the extra data within mjpg-streamer?

Did you try -y?