jacksonliam/mjpg-streamer #238: mjpeg-streamer Docker use /dev/video0 instead the defined device (/dev/video2)

redskinhu, 2020-01-26:
Hello

I found a small issue. I have 2 same type of USB cameras connected to the machine.
If I use command line mjpg-streamer the cams are working fine.

Some details:

```
Microsoft® LifeCam HD-3000: Mi (usb-0000:00:14.0-10.3):
	/dev/video2
	/dev/video3

Microsoft® LifeCam HD-3000: Mi (usb-0000:00:14.0-11.4):
	/dev/video0
	/dev/video1
```

/dev/video0  on 8080 port
/dev/video2  on 8082 port

If I wanted to use the mjpg-streamer with Docker the first cam (/dev/video0) working properly but I can't start the second cam (/dev/video2), even if the first not started/created. The mjpg-sreamer wanted to use the /video/dev0 device, see the rows 8-9. Why? :-) 

This is the error message:

```
MJPG-streamer [7]: starting application
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
MJPG-streamer [7]: MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912

 i: Using V4L2 device.: /dev/video0
MJPG-streamer [7]: Using V4L2 device.: /dev/video0

 i: Desired Resolution: 640 x 480
MJPG-streamer [7]: Desired Resolution: 640 x 480

 i: Frames Per Second.: 30
MJPG-streamer [7]: Frames Per Second.: 30

 i: Format............: JPEG
MJPG-streamer [7]: Format............: JPEG

 i: TV-Norm...........: DEFAULT
MJPG-streamer [7]: TV-Norm...........: DEFAULT

ERROR opening V4L interface: No such file or directory
 i: init_VideoIn failed
MJPG-streamer [7]: init_VideoIn failed
```
I use modified Dockerfile (different port):

```
FROM chunliu/docker-opencv
MAINTAINER Chun Liu <https://github.com/chunliu>
LABEL Description="A Docker image for mjpg_streamer." Version="0.3"

# Build mjpg_streamer
# Need to run container with the device: docker run -t -i -p 8080:8080/tcp --device=/dev/video0 image:tag
RUN git clone https://github.com/jacksonliam/mjpg-streamer.git 

WORKDIR /mjpg-streamer/mjpg-streamer-experimental

RUN make \ 
    && make install \
    && chmod +x docker-start.sh

EXPOSE 8082/TCP

ENTRYPOINT ["/mjpg-streamer/mjpg-streamer-experimental/docker-start.sh", "output_http.so -n"]

CMD ["input_uvc.so -r 640x480 -f 30 -q 100 -n"]

```
And this is the way how I created the Container (different port and dev):
`sudo docker run -t -i -p 8082:8082/tcp --device=/dev/video2 mjpg-streamer_8082:latest`

Is it possible to fix it somehow?

Thx
T

jacksonliam, 2020-01-26:
Hey, I didn't write the docker stuff so not too sure.

I guess the --device arg is not being passed to mjpg-streamer correctly. Possibly something to do with the docker_start.sh script.

redskinhu, 2020-01-27:
Solution:

You need to define the port and the video device too in the Dockerfile:
If you don't the mjpg streamer starts with the default /dev/video0 and the 8080 port value...

```
FROM chunliu/docker-opencv
MAINTAINER Chun Liu <https://github.com/chunliu>
LABEL Description="A Docker image for mjpg_streamer." Version="0.3"
RUN git clone https://github.com/jacksonliam/mjpg-streamer.git 
WORKDIR /mjpg-streamer/mjpg-streamer-experimental
RUN make \ 
    && make install \
    && chmod +x docker-start.sh
EXPOSE 8082/TCP
ENTRYPOINT ["/mjpg-streamer/mjpg-streamer-experimental/docker-start.sh", "output_http.so -n -p 8082"]
CMD ["input_uvc.so -d /dev/video2 -r 640x480 -f 30 -q 100 -n"]
```

`sudo docker run -t -i -p 8082:8082/tcp --device=/dev/video2 mjpg-streamer:latest`

closed 2020-01-27 by redskinhu