jacksonliam/mjpg-streamer #239: Cleaning up resources allocated by server thread - Hang

victorromeo, 2020-02-09:
When I create a stream using input_file, and I issue a termination using ctrl-c combination on the keyboard, the service hangs rather than closes.  Thoughts?

```
mjpg_streamer -o 'output_http.so -w ./www' -i 'input_file.so -f /home/pi/Service/capture/photo'
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: folder to watch...: /home/pi/Service/capture/photo/
 i: forced delay......: 1.0000
 i: delete file.......: no, do not delete
 i: filename must be..: -no filter for certain filename set-
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
q
^Csetting signal to stop
force cancellation of threads and cleanup resources
 o: cleaning up resources allocated by server thread #00
```

victorromeo, 2020-02-09:
On "uname -a"
Linux agtech01 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

victorromeo, 2020-02-09:
Doesn't seem to occur if the server is closed soon after starting (30 seconds)

bpengu1n, 2020-05-08:
This is related to/referenced by issue #182 - the solution referenced there was to migrate to a different application. Unfortunately not really a solution for those of us who want to keep using this one...

bvitaz-ck, 2022-03-01:
I am experiencing this as well.