jacksonliam/mjpg-streamer #240: Remote control for input_raspicam.so do not working

bdn76, 2020-02-11:
Rasbian stable, RPi 3B+, commit 501f6362c5afddcfb41055f97ae484252c85c912. Streamer starting with:

/usr/local/bin/mjpg_streamer -o "output_http.so -w /usr/local/share/mjpg-streamer/www" -i "input_raspicam.so -x 1920 -y 1080 -ISO 32000"

but

[image]

snapshot/stream work fine.

jacksonliam, 2020-02-11:
Could you elaborate? I'm not really sure what your problem is here...

bdn76, 2020-02-11:
If i change input to input_uvc.so with -d /dev/video0 then control page appear but i'm unable to control camera but with input_raspicam.so control page is empty.

jacksonliam, 2020-02-11:
This is expected, the control page is only implemented for uvc cameras. It is not supported for input_raapicam. Would be interesting to implement it, but unfortunately I do not have the time!

closed 2020-02-11 by jacksonliam