jacksonliam/mjpg-streamer #241: RE: libjpeg8-dev/Linux Debian 4.14.x

silver2row, 2020-02-29:
Hello,

I have a Linux sbc and I am trying to make this source work for me and a camera attachment.

...

Um, libjpeg8-dev is not any longer available on my debian distro. Is there any way around this portion of the source? Also, I have gcc 6.3.0 and g++ 6.3.0.

I was wondering if the combination of different c/c++ compilation tools takes away from the workings of your Makefiles.

Seth

P.S. I am sure it does but is there a way to make the compilation work w/ my versioned tools and the libjpeg62-turbo-dev instead?

silver2row, 2020-02-29:
Oh...if you need any compilation errors, please let me know.

Seth

jacksonliam, 2020-02-29:
Yes it should work with other versions of libjpeg, not sure about the turbo one.

What are the compile errors?

silver2row, 2020-03-01:
Hello,

There were too many to count or list here. I will try again w/ a fresh install. 

Seth

P.S. I will report back here in a couple of days. Thank you for answering so quickly.

silver2row, 2020-03-01:
Hello @jacksonliam ,

Seth here. Look here:

sudo apt install cmake libjpeg8-dev
    Reading package lists... Done
    Building dependency tree
    Reading state information... Done
    Package libjpeg8-dev is not available, but is referred to by another package.
    This may mean that the package is missing, has been obsoleted, or
    is only available from another source
    However the following packages replace it:
      libjpeg62-turbo-dev

    E: Package 'libjpeg8-dev' has no installation candidate

That is what prints out to console when I try to install libjpeg8-dev w/ apt on debian 4.14.x.

Seth

P.S. Please reply when you have some extra minutes.

silver2row, 2020-03-02:
Hello @jacksonliam ,

I got the source to compile when I erased the input_opencv directory and changed the CMakeLists.txt file to suit my needs.

Still, I cannot view the feed of my camera when I run the basic operation command:

    ./mjpg_streamer -o "output_http.so -w ./www"

Seth

P.S. Do you have any ideas on why I cannot view my feed? I can get to the site via that command and <my IP>:8080 in the browser.

jacksonliam, 2020-03-02:
You're not specifying an input plugin?

silver2row, 2020-03-02:
Hello,

Oh. Okay. So, I need to supply an input plugin. Okay. So, something like this would work (maybe)?

    ./mjpg_streamer -i "input_http.so" -o "output_http.so -w ./www"

Seth

P.S. I will test more. I just erased a bunch of items on my board b/c of the port 8080 being taken up. So, testing will ensue. Thank you, sir.

closed 2020-03-02 by silver2row