jacksonliam/mjpg-streamer #243: Make issue

CastillloFrancisco2018, 2020-04-14:
after running the make  mjpg_streamer input_file.so input_uvc.so output_http.so command i get this:

cc     mjpg_streamer.c   -o mjpg_streamer
/usr/bin/ld: /tmp/ccS5nJFg.o: in function `signal_handler':
mjpg_streamer.c:(.text+0x3b8): undefined reference to `dlclose'
/usr/bin/ld: mjpg_streamer.c:(.text+0x4e4): undefined reference to `dlclose'
/usr/bin/ld: /tmp/ccS5nJFg.o: in function `main':
mjpg_streamer.c:(.text+0xa10): undefined reference to `daemon_mode'
/usr/bin/ld: mjpg_streamer.c:(.text+0xe1c): undefined reference to `dlopen'
/usr/bin/ld: mjpg_streamer.c:(.text+0xfd8): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1064): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x10cc): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1154): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x11bc): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1244): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x12ac): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1334): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1738): undefined reference to `dlopen'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1954): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x19f8): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1a88): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1b28): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1bc0): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1c60): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1cf8): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1d98): undefined reference to `dlsym'
collect2: error: ld returned 1 exit status
make: *** [<builtin>: mjpg_streamer] Error 1

im not sure where to go from here...

jacksonliam, 2020-04-14:
Try running `LDFLAGS='-ldl'` before make

CastillloFrancisco2018, 2020-04-14:
i ran: LDLFLAGS='ldl'make mjpg_streamer input_file.so input_uvc.so output_http.so

and i got:   -bash: mjpg_streamer: command not found

CastillloFrancisco2018, 2020-04-14:
if i run:   LDLFLAGS='ldl' make mjpg_streamer input_file.so input_uvc.so output_http.so
with a space between ldlflags and make it just gives the same output as the first time

cc     mjpg_streamer.c   -o mjpg_streamer
/usr/bin/ld: /tmp/ccNM7syA.o: in function `signal_handler':
mjpg_streamer.c:(.text+0x3b8): undefined reference to `dlclose'
/usr/bin/ld: mjpg_streamer.c:(.text+0x4e4): undefined reference to `dlclose'
/usr/bin/ld: /tmp/ccNM7syA.o: in function `main':
mjpg_streamer.c:(.text+0xa10): undefined reference to `daemon_mode'
/usr/bin/ld: mjpg_streamer.c:(.text+0xe1c): undefined reference to `dlopen'
/usr/bin/ld: mjpg_streamer.c:(.text+0xfd8): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1064): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x10cc): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1154): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x11bc): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1244): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x12ac): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1334): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1738): undefined reference to `dlopen'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1954): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x19f8): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1a88): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1b28): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1bc0): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1c60): undefined reference to `dlsym'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1cf8): undefined reference to `dlerror'
/usr/bin/ld: mjpg_streamer.c:(.text+0x1d98): undefined reference to `dlsym'
collect2: error: ld returned 1 exit status
make: *** [<builtin>: mjpg_streamer] Error 1

CastillloFrancisco2018, 2020-04-14:
this is everything in the mjpg-streamer-experimental folder
cmake                  LICENSE                  mjpg_streamer.h                   README.md     utils.c
CMakeLists.txt     makedeb.sh            mjpg_streamer@.service       scripts               utils.h
Dockerfile            Makefile                  plugins                                  start.sh              www
docker-start.sh    mjpg_streamer.c      postinstall.sh                        TODO

CastillloFrancisco2018, 2020-04-14:
this is everything in the mjpg-experimental-folder:
cmake            
LICENSE          
mjpg_streamer.h         
README.md  
utils.c
CMakeLists.txt   
makedeb.sh       
mjpg_streamer@.service  
scripts    
utils.h
Dockerfile       
Makefile         
plugins                 
start.sh   
www
docker-start.sh  
mjpg_streamer.c  
postinstall.sh          
TODO

jacksonliam, 2020-04-14:
Hang on are you running the command:
`make mjpg_streamer input_file.so input_uvc.so output_http.so`
???

That is not valid. Where told you to do that?

CastillloFrancisco2018, 2020-04-14:
im following this video on youtube:
https://www.youtube.com/watch?v=xt-rrH04O2g

jacksonliam, 2020-04-15:
Ok I see, it's 5 years old... Still I'd rather the instructor wouldn't have given that instruction, even 5 year ago!

Anyway... Delete the mjpg-streamer directory and clone from git again because I don't know if this will have created some files which will mess things up.

cd into the mjpg-streamer-experimental directory again and this time just run `make` on its own. No other arguments.

closed 2020-04-15 by jacksonliam

CastillloFrancisco2018, 2020-04-15:
i got this:
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
/bin/sh: 1: cmake: not found
make: *** [Makefile:18: all] Error 127