jacksonliam/mjpg-streamer #246: After unplugging the webcam, mjpeg-streamer fails to start until OS restarts.

poikilos, 2020-04-18:
I have looked at other issues, but I can't say for sure it is related to any of them. I am not sure whether
- mjpeg-streamer can somehow detect something differently
- ... must control the video device
- whether there is actually a bug in v4l
- whether restarting v4l is possible or necessary

## Actual behavior
- Disconnect the webcam from the USB port.
- Reconnect the webcam from the USB port (either the same USB port or a different one).
- (mjpeg-streamer will not run until the next restart of the OS.)

## Expected behavior
- The same command works even after the webcam is disconnected and reconnected.

## Environment
- OS: Debian 10 (I made a systemd unit to run the webcam as the same unprivileged user as Octoprint, `pi` in my case).
- Webcam: (USB) "GUCEE HD92 1080p" is printed on its mount. It is a generic webcam, listed at time of purchase (but now [link is dead) as "2MP Webcam 1080P, GUCEE HD92 Full HD Web Camera 110 Degree Wide Angle Crystal Clear Video with Noise Cancelling Mic, Skype Webcam for PC, Mac, Laptop](https://www.amazon.com/gp/product/B06Y5K6RKY/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1&tag=chromeapp02-21)," (and other names) on Amazon.

## Details
<details>
  <summary>Here is the relevant output of `journalctl -xe` (which may include output from more than one attempt at `systemctl restart mjpeg-streamer`)</summary>
```
Apr 18 16:14:57 buzzy bash[19998]:  i: Frames Per Second.: -1
Apr 18 16:14:57 buzzy bash[19998]:  i: Format............: JPEG
Apr 18 16:14:57 buzzy bash[19998]:  i: TV-Norm...........: DEFAULT
Apr 18 16:14:57 buzzy bash[19998]:  i: init_VideoIn failed
Apr 18 16:14:57 buzzy mjpg_streamer[19999]: MJPG-streamer [19999]: Using V4L2 device.: (null)
Apr 18 16:14:57 buzzy systemd[1]: mjpeg-streamer.service: Main process exited, code=exited, status=1/FAILURE
-- Subject: Unit process exited
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- An ExecStart= process belonging to unit mjpeg-streamer.service has exited.
-- 
-- The process' exit code is 'exited' and its exit status is 1.
Apr 18 16:14:57 buzzy mjpg_streamer[19999]: MJPG-streamer [19999]: Desired Resolution: 1280 x 720
Apr 18 16:14:57 buzzy systemd[1]: mjpeg-streamer.service: Failed with result 'exit-code'.
-- Subject: Unit failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- The unit mjpeg-streamer.service has entered the 'failed' state with result 'exit-code'.
Apr 18 16:14:57 buzzy mjpg_streamer[19999]: MJPG-streamer [19999]: Frames Per Second.: -1
Apr 18 16:14:57 buzzy mjpg_streamer[19999]: MJPG-streamer [19999]: Format............: JPEG
Apr 18 16:14:57 buzzy mjpg_streamer[19999]: MJPG-streamer [19999]: TV-Norm...........: DEFAULT
Apr 18 16:14:57 buzzy mjpg_streamer[19999]: MJPG-streamer [19999]: init_VideoIn failed
```
</details>

The service file runs my script as follows:
```
[Unit]
Description=mjpeg-streamer service (as pi).

[Service]
Type=simple
ExecStart=/bin/bash /home/pi/mjpeg-streamer.sh
User=pi
Group=pi

[Install]
WantedBy=multi-user.target
```

The script contains:
```bash
#!/bin/sh
LD_LIBRARY_PATH=/home/pi/mjpg-streamer/mjpg-streamer-experimental /home/pi/mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer -i "input_uvc.so -d /dev/video0 -r 1280x720" -o "output_http.so"
```

If I manually run `su pi` then the command above, I get:
```
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using V4L2 device.: (null)
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: init_VideoIn failed
```

Restarting the OS solves the problem, resulting in:
```
# systemctl status mjpeg-streamer.service
● mjpeg-streamer.service - mjpeg-streamer service (as pi).
   Loaded: loaded (/etc/systemd/system/mjpeg-streamer.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-04-18 16:38:40 EDT; 2min 12s ago
 Main PID: 281 (bash)
    Tasks: 4 (limit: 4915)
   Memory: 3.7M
   CGroup: /system.slice/mjpeg-streamer.service
           ├─281 /bin/bash /home/pi/mjpeg-streamer.sh
           └─332 /home/pi/mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer -i input_uvc.so -d /dev/video0 -r 1280x720 -o output_http.so

Apr 18 16:38:41 buzzy bash[281]:  o: HTTP Listen Address..: (null)
Apr 18 16:38:41 buzzy bash[281]:  o: username:password....: disabled
Apr 18 16:38:41 buzzy bash[281]:  o: commands.............: enabled
Apr 18 16:38:41 buzzy mjpg_streamer[332]: MJPG-streamer [332]: www-folder-path......: disabled
Apr 18 16:38:41 buzzy mjpg_streamer[332]: MJPG-streamer [332]: HTTP TCP port........: 8080
Apr 18 16:38:41 buzzy mjpg_streamer[332]: MJPG-streamer [332]: HTTP Listen Address..: (null)
Apr 18 16:38:41 buzzy mjpg_streamer[332]: MJPG-streamer [332]: username:password....: disabled
Apr 18 16:38:41 buzzy mjpg_streamer[332]: MJPG-streamer [332]: commands.............: enabled
Apr 18 16:38:41 buzzy mjpg_streamer[332]: MJPG-streamer [332]: starting input plugin input_uvc.so
Apr 18 16:38:41 buzzy mjpg_streamer[332]: MJPG-streamer [332]: starting output plugin: output_http.so (ID: 00)
```

jacksonliam, 2020-04-19:
Is mjpg-streamer definitely not already running when you try to start it again after plugging the camera back in?

If so, I would think this is a v4l2 or camera driver issue. Unless there is something mjpg-streamer is not cleaning up when it exits.

I don't think I can reproduce this my side, but will try to check with my USB camera if I get time.

poikilos, 2020-04-22:
OctoPrint can't access it during the times I describe that mjpg-streamer is not working, so that seems to indicate for sure that it isn't running.

jacksonliam, 2020-04-22:
Check via `ps -ef` to see what's running. It may be as simple as killing it then starting it again.

poikilos, 2020-12-19:
I don't understand why you closed this. Why should the daemon die when the camera is unplugged and not come back automatically? Isn't there a solution?

jacksonliam, 2020-12-19:
Closed due to lack of response. 
This seems to be a problem specific to your system/camera. It is not something I can reproduce unfortunately.

I'm not even sure it is an mjpg-streamer problem. Unless there is anything to suggest otherwise (e.g. other v4l2 software can continue to capture after the camera is unplugged/replugged) I'm not sure what I can fix here?

closed 2020-12-19 by jacksonliam

poikilos, 2021-01-07:
> Closed due to lack of response.
> This seems to be a problem specific to your system/camera. It is not something I can reproduce unfortunately.
>
> I'm not even sure it is an mjpg-streamer problem. [...]

Yes, ffmpeg can do it: `ffmpeg -f v4l2 -video_size 1280x720 -i /dev/video1 -frames 1 video1.jpg`
- works and the image is correct (I can't test using OBS or other GUI programs because the machine is a minimal headless Debian server and has a small flash hard drive--feel free to suggest one).

poikilos, 2021-01-07:
> Check via `ps -ef` to see what's running. It may be as simple as killing it then starting it again.

`systemctl restart mjpeg-streamer` doesn't solve the issue--is that enough to perform that test?

ILAsoft, 2021-01-29:
FWIW I am getting the same exact behavior - upon reconnect, while dmesg shows the camera connected fine, the streamer (and thus my octoprint) does not pick it up. Running "`sudo service webcamd restart`" fixes the issue immediately (!). Also note that I see that on every reconnect, the device input path is different (increases), e.g.:
```
[Thu Jan 28 20:19:31 2021] input: USB2.0 PC CAMERA: USB2.0 PC CAM as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/input/input3
[.......]
[Thu Jan 28 20:19:33 2021] input: USB2.0 PC CAMERA: USB2.0 PC CAM as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/input/input4
```

ILAsoft, 2021-01-30:
Sorry for the SPAM, but figured I'd share some extra info in case it helps -- this is the status once the camera "dies" and gets successfully reconnected thereafter, but the service does not pick it up, forcing me to issue the restart:
`service webcamd status
● webcamd.service - the OctoPi webcam daemon with the user specified config
   Loaded: loaded (/etc/systemd/system/webcamd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-01-30 04:42:13 PST; 9h ago
  Process: 17602 ExecStart=/root/bin/webcamd (code=exited, status=0/SUCCESS)
 Main PID: 17627 (mjpg_streamer)
    Tasks: 7 (limit: 1939)
   CGroup: /system.slice/webcamd.service
           └─17627 ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_uvc.so -r 1280x720 -f 10 -d /dev/video0

Jan 30 04:42:13 octopi mjpg_streamer[17627]: MJPG-streamer [17627]: starting output plugin: output_http.so (ID: 00)
Jan 30 04:42:13 octopi webcamd[17602]: Done bring up all configured video device
Jan 30 04:42:13 octopi webcamd[17602]: Goodbye...
Jan 30 04:42:13 octopi systemd[1]: Started the OctoPi webcam daemon with the user specified config.
Jan 30 13:49:51 octopi webcamd[17602]: Unable to dequeue buffer: No such device
Jan 30 13:49:51 octopi webcamd[17602]:  i: Error grabbing frames
Jan 30 13:49:51 octopi webcamd[17602]:  i: cleaning up resources allocated by input thread
Jan 30 13:49:51 octopi webcamd[17602]: Unable to stop capture: No such device
Jan 30 13:49:51 octopi mjpg_streamer[17627]: MJPG-streamer [17627]: Error grabbing frames
Jan 30 13:49:51 octopi mjpg_streamer[17627]: MJPG-streamer [17627]: cleaning up resources allocated by input thread`

Side note - my temporary solution is a basic `journalctl -u webcamd -b -n 1 | grep cleaning && sudo service webcamd restart` in crontab for now...

username7291, 2022-11-12:
>This seems to be a problem specific to your system/camera. It is not something I can reproduce unfortunately.
Have the same on raspberry with pure mjpg_streamer. Unplug the cam while mjpg_streamer is running. It cant closed with CTRL-C, using kill dont help cam unavailable until restart the whole system. Maybe VideoX is blocked, cause not really free´d and cant recreated if the cam is plugged again.

poikilos, 2022-11-21:
@username7291 You have to put a blank line after the quote, or what you are saying will be included in the blockquote.

Poikilos, 2023-01-22:
`systemctl restart mjpeg-streamer` works now after unplugging and re-plugging the same webcam. I am now on mjpg-streamer-experimental 310b29f4a94c46652b20c4b7b6e5cf24e532af39.

It is solved apparently, because besides updating mjpeg-streamer nothing known to be significant changed (still Debian 10, but updated packages).