jacksonliam/mjpg-streamer #248: RGB3 support?

GilesBathgate, 2020-04-24:
When I run mjpg-streamer I get the following output:

```
./mjpg_streamer -i 'input_uvc.so -r 352x288'
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 352 x 288
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: S910
    ... will try to handle this by checking against supported formats.
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
```

If I enable v4l1compat support I get this:

```
LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so ./mjpg_streamer -i 'input_uvc.so -r 352x288'
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 352 x 288
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: RGB3
    ... will try to handle this by checking against supported formats.
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
```

GilesBathgate, 2020-04-24:
```
v4l2-ctl --all
Driver Info (not using libv4l2):
        Driver name   : sonixb
        Card type     : USB camera
        Bus info      : usb-0000:00:1d.0-1
        Driver version: 4.9.30
        Capabilities  : 0x85200001
                Video Capture
                Read/Write
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x05200001
                Video Capture
                Read/Write
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (sonixb: ok)
Format Video Capture:
        Width/Height      : 352/288
        Pixel Format      : 'RGB3'
        Field             : None
        Bytes per Line    : 1056
        Size Image        : 304128
        Colorspace        : sRGB
        Transfer Function : Default
        YCbCr/HSV Encoding: Default
        Quantization      : Default
        Flags             :
Streaming Parameters Video Capture:
        Frames per second: invalid (0/0)
        Read buffers     : 2

User Controls

                       exposure (int)    : min=2 max=15 step=1 default=2 value=15 flags=inactive
                 gain_automatic (bool)   : default=1 value=1 flags=update
                           gain (int)    : min=0 max=255 step=1 default=127 value=218 flags=inactive

```

GilesBathgate, 2020-04-24:
```
v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'RGB3' (emulated)
        Name        : RGB3
                Size: Discrete 160x120
                Size: Discrete 176x144
                Size: Discrete 320x240
                Size: Discrete 352x288

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'BGR3' (emulated)
        Name        : BGR3
                Size: Discrete 160x120
                Size: Discrete 176x144
                Size: Discrete 320x240
                Size: Discrete 352x288

        Index       : 2
        Type        : Video Capture
        Pixel Format: 'YU12' (emulated)
        Name        : YU12
                Size: Discrete 160x120
                Size: Discrete 176x144
                Size: Discrete 320x240
                Size: Discrete 352x288

        Index       : 3
        Type        : Video Capture
        Pixel Format: 'YV12' (emulated)
        Name        : YV12
                Size: Discrete 160x120
                Size: Discrete 176x144
                Size: Discrete 320x240
                Size: Discrete 352x288
```

jacksonliam, 2020-04-24:
Yeah we only support RGBP, don't see many RGB only cameras tbh. They've normally got YUV.

I've seen a few that only support YU12 or YV12 like yours. Would like to add support for those but won't have time to do it myself.

closed 2020-04-24 by jacksonliam

GilesBathgate, 2020-04-24:
@jacksonliam any idea how I can even switch to the YU12 or YV12 modes? also, could I hack a file-based mode if I use mencoder to output files.

Armageddit, 2020-04-25:
same problem with old logitech webcam

GilesBathgate, 2020-04-25:
@Armageddit Can you post some details such as the output of `v4l2-ctl --list-formats-ext` to see if my fix will help.

Armageddit, 2020-04-29:
oh sry i think it was RGB3 becaus if i try to start:

i type in:
```
cd mjpg-streamer/mjpg-streamer-experimental
./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"
```

and get:

```
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: The specified resolution is unavailable, using: width 352 height 288 instead
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: RGB3
    ... will try to handle this by checking against supported formats.
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
```

here the results from `v4l2-ctl --list-formats-ext`

```
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'BA81'
        Name        : BA81
                Size: Discrete 176x144
                Size: Discrete 352x288
```

i try the second camera that works only for 10sec and then i must restart the programm

```
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'JPEG' (compressed)
        Name        : JPEG
                Size: Discrete 320x232
                Size: Discrete 640x472
```

im not good with linux or terminal  can you tell me how to get this work?
i just folow this instructions to run on my bananapi:

[https://community.octoprint.org](https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337)

GilesBathgate, 2020-04-29:
@Armageddit This could be quite tricky to get it to work, unless you are happy compiling mjpg-streamer from source yourself. If not you will have to wait for @jacksonliam to accept my fix and issue a new release.

That said, there are a few extra things to test. See if you can find the `v4l1compat.so` file:

`sudo find / -name "v4l1compat.so"`

It might give you something like `/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so`
if you find then use the full path to try loading it:

`export LD_PRELOAD=`PUT_PATH_HERE`/v4l1compat.so`

Then try running your tests as before again.

Armageddit, 2020-04-29:
ok i run:

```
Linux lemaker 3.4.103 #4 SMP PREEMPT Thu Dec 18 12:55:58 CST 2014 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
bananapi@lemaker ~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
bananapi@lemaker ~ $ sudo find / -name "v4l1compat.so"
[sudo] Passwort für bananapi:
find: ‘/run/user/1000/gvfs’: Keine Berechtigung
/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so
bananapi@lemaker ~ $ export LD_PRELOAD= /usr/lib/arm-linux-gnueabihf/libv4l/v4l1                  compat.so
-bash: export: `/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so': Ist kein gül                  tiger Bezeichner.
bananapi@lemaker ~ $ export LD_PRELOAD= /usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so
-bash: export: `/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so': Ist kein gültiger Bezeichner.
bananapi@lemaker ~ $ export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so
bananapi@lemaker ~ $ cd mjpg-streamer/mjpg-streamer-experimental
bananapi@lemaker ~/mjpg-streamer/mjpg-streamer-experimental $ ./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: The specified resolution is unavailable, using: width 352 height 288 instead
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: RGB3
    ... will try to handle this by checking against supported formats.
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
bananapi@lemaker ~/mjpg-streamer/mjpg-streamer-experimental $ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'RGB3' (emulated)
        Name        : RGB3
                Size: Discrete 176x144
                Size: Discrete 352x288

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'BGR3' (emulated)
        Name        : BGR3
                Size: Discrete 176x144
                Size: Discrete 352x288

        Index       : 2
        Type        : Video Capture
        Pixel Format: 'YU12' (emulated)
        Name        : YU12
                Size: Discrete 176x144
                Size: Discrete 352x288

        Index       : 3
        Type        : Video Capture
        Pixel Format: 'YV12' (emulated)
        Name        : YV12
                Size: Discrete 176x144
                Size: Discrete 352x288

bananapi@lemaker ~/mjpg-streamer/mjpg-streamer-experimental $
```

GilesBathgate, 2020-04-29:
You need `export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so` without the space. However It does look like `v4l2ctl --list-formats-ext` is working as expected now, and outputs the RGB3 format. My change should fix this. Do you want to try compiling yourself, I can give you instructions, but they are in the https://github.com/jacksonliam/mjpg-streamer/blob/master/README.md

GilesBathgate, 2020-04-29:
You will need to `git clone git@github.com:GilesBathgate/mjpg-streamer.git`, and then `git checkout rgb24-format-support`

Armageddit, 2020-04-29:
it is not empty:
```
bananapi@lemaker ~ $ git clone git@github.com:GilesBathgate/mjpg-streamer.git
fatal: Zielpfad 'mjpg-streamer' existiert bereits und ist kein leeres Verzeichnis.
bananapi@lemaker ~ $ git checkout rgb24-format-support
fatal: Kein Git-Repository (oder irgendein Elternverzeichnis): .git
```
i try to understand the README.md but dont understand what i should do

GilesBathgate, 2020-04-29:
Sorry unless you have set up in your GitHub account, you need to clone via https:

`git clone https://github.com/GilesBathgate/mjpg-streamer.git`
then change to the folder
`cd mjpg-streamer`
then 
`git checkout rgb24-format-support`

Armageddit, 2020-04-29:
```
bananapi@lemaker ~ $ git clone git@github.com:GilesBathgate/mjpg-streamer.git
Klone nach 'mjpg-streamer' ...
Permission denied (publickey).
fatal: Konnte nicht vom Remote-Repository lesen.

Bitte stellen Sie sicher, dass die korrekten Zugriffsberechtigungen bestehen
und das Repository existiert.

```

GilesBathgate, 2020-04-29:
You've done the same thing! NOT `git@github.com...` YOU NEED `https://github.com...`

Armageddit, 2020-04-29:
ok an now:
```
export LD_LIBRARY_PATH=.
make
```

?

GilesBathgate, 2020-04-29:
No

>Simple compilation
This will build and install all plugins that can be compiled.

`cd mjpg-streamer-experimental`
and then
`make`

Armageddit, 2020-04-29:
oh yes :D
BIG THX!!!
it works i see the videostream :D

GilesBathgate, 2020-04-29:
Cool can you place screenshot on #249

Armageddit, 2020-04-30:
after restart i need this 3 codes to start:

```
export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so
cd mjpg-streamer/mjpg-streamer-experimental 
 ./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"
```

```
bananapi@lemaker ~ $ export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1c             ompat.so
bananapi@lemaker ~ $ cd mjpg-streamer/mjpg-streamer-experimental                             bananapi@lemaker ~/mjpg-streamer/mjpg-streamer-experimental $ ./mjpg_streamer -i              "./input_uvc.so" -o "./output_http.so"
MJPG Streamer Version: git rev: f708ce284183fad083511383978460b3018943eb
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: The specified resolution is unavailable, using: width 352 height 288 instead             
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: RGB3
    ... will try to handle this by checking against supported formats.
    ... Falling back to RGB24 mode (consider using -fourcc RGB24 option). Note t             hat this requires much more CPU power
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for dev             ice (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (2             5)
 o: www-folder-path......: disabled
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
```

GilesBathgate, 2020-04-30:
Firstly, if you run `sudo make install` from the `~/mjpg-streamer/mjpg-streamer-experimental` directory it will install it so you can run it from any location.

If you want to automate the startup you can add to systemd:

`/etc/systemd/system/mjpg-streamer.service`
Create the file:
```
[Unit]
Description=Streams video with Raspberry Pi Camera
After=syslog.target
After=network.target
Requires=dev-video0.device
BindsTo=dev-video0.device

[Service]
ExecStart=/usr/local/bin/mjpg_streamer -i "/usr/local/lib/mjpg-streamer/input_uvc.so -r 352x288 -fourcc RGB24"
Environment="LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so"

[Install]
WantedBy=multi-user.target
```
You will have to update ENVIRONMENT to your setting.

GilesBathgate, 2020-04-30:
After completing the above test the service with
`sudo systemctl start mjpg-streamer`
if the camera is now working use:
`sudo systemctl enable mjpg-streamer`
then it will work after restart.

Armageddit, 2020-05-01:
dont work from any location:

```
bananapi@lemaker ~ $ export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so
bananapi@lemaker ~ $ ./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"
-bash: ./mjpg_streamer: Datei oder Verzeichnis nicht gefunden
bananapi@lemaker ~ $
```

and this file doesnt work too :(
```
[Unit]
Description=Streams video with Raspberry Pi Camera
After=syslog.target
After=network.target
Requires=dev-video0.device
BindsTo=dev-video0.device

[Service]
ExecStart=/usr/local/bin/mjpg_streamer -i "/usr/local/lib/mjpg-streamer/input_uvc.so -r 352x288 -fourcc RGB24"
Environment="LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so"

[Install]
WantedBy=multi-user.target
```
i try to start wiht `sudo systemctl start mjpg-streamer`
no response and dont work
even after a restart

but if i use this 3 commands it woke fine:

```
export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so
cd mjpg-streamer/mjpg-streamer-experimental 
 ./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"
```

it is possible that it was installed in the wrong directory

Armageddit, 2020-05-01:
```
bananapi@lemaker /usr/local/bin $ ls
bp-update  ffmpeg  ffprobe  gpio  indiecity  pip  pip2  pip2.7  wheel
```

GilesBathgate, 2020-05-01:
When you add `./` it means run from current directory. So instead of `./mjpg-streamer` you need just `mjpg-streamer`. Regarding the install location yes you will need to adjust it to the correct location `which mjpg-streamer` will tell you where it is located.

Armageddit, 2020-05-04:
ok now it works fine :)
i have done all again and now it starts automatic and can stop with `sudo systemctl stop mjpg-streamer` or dont startup atomatic with `sudo systemctl disable mjpg-streamer`

BIG THX TO GilesBathgate!!

fabiosci, 2022-02-22:
Hi,
i've found this thread while trying to use an old Logitech webcam with octoprint. I followed all steps and when i execute the following commands one by one everything work as expected:
```
export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so
/usr/local/bin/mjpg_streamer -i "/home/pi/mjpg-streamer/input_uvc.so -r 352x288"
```

More precisely, the output are:
```
pi@octopi:~ $ export
declare -x DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"
declare -x HOME="/home/pi"
declare -x LANG="en_GB.UTF-8"
declare -x LD_PRELOAD="/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so"
declare -x LOGNAME="pi"
declare -x LS_COLORS="rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:"
declare -x MAIL="/var/mail/pi"
declare -x OLDPWD
declare -x PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games"
declare -x PWD="/home/pi"
declare -x SHELL="/bin/bash"
declare -x SHLVL="1"
declare -x SSH_CLIENT="192.168.188.37 50934 22"
declare -x SSH_CONNECTION="192.168.188.37 50934 192.168.188.39 22"
declare -x SSH_TTY="/dev/pts/0"
declare -x TERM="xterm"
declare -x TEXTDOMAIN="Linux-PAM"
declare -x USER="pi"
declare -x XDG_RUNTIME_DIR="/run/user/1000"
declare -x XDG_SESSION_CLASS="user"
declare -x XDG_SESSION_ID="1"
declare -x XDG_SESSION_TYPE="tty"
```
and:
```
pi@octopi:~ $ /usr/local/bin/mjpg_streamer -i "/home/pi/mjpg-streamer/input_uvc.so -r 352x288"
MJPG Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 352 x 288
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: RGB3
    ... will try to handle this by checking against supported formats.
    ... Falling back to RGB24 mode (consider using -fourcc RGB24 option). Note that this requires much more CPU power
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: disabled
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled

```
and everything is ok in octoprint.

[... 13 lines cut ...]
[Install]
WantedBy=multi-user.target
``` 

```
pi@octopi:~ $ systemctl status mjpg-streamer
● mjpg-streamer.service - Streams video with Raspberry Pi Camera
   Loaded: loaded (/etc/systemd/system/mjpg-streamer.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Tue 2022-02-22 22:22:56 GMT; 5s ago
  Process: 22310 ExecStart=/usr/local/bin/mjpg_streamer -i /home/pi/mjpg-streamer/input_uvc.so -r 352x288 (code=killed, signal=TERM)
 Main PID: 22310 (code=killed, signal=TERM)

Feb 22 22:22:56 octopi systemd[1]: Started Streams video with Raspberry Pi Camera.
[... cut ...]
Feb 22 22:22:56 octopi systemd[1]: Stopping Streams video with Raspberry Pi Camera...
Feb 22 22:22:56 octopi systemd[1]: mjpg-streamer.service: Main process exited, code=killed, status=15/TERM
Feb 22 22:22:56 octopi systemd[1]: mjpg-streamer.service: Succeeded.
Feb 22 22:22:56 octopi systemd[1]: Stopped Streams video with Raspberry Pi Camera.

```

I tried adding `After=dev-video0.device` and using `sudo systemctl daemon-reload` on enabled service without success.

Any suggestion to make it work is very appreciated.
many thanks

GilesBathgate, 2022-02-22:
> Any suggestion to make it work is very appreciated. many thanks

First step might be to use journalctl to debug the messages from mjpg_streamer?

fabiosci, 2022-02-23:
sorry, forgot to mention that i tried with journalctl

```
pi@octopi:~ $ sudo journalctl | grep mjpg-streamer
Feb 23 09:51:52 octopi systemd[1]: mjpg-streamer.service: Unit is bound to inactive unit dev-video0.device. Stopping, too.
Feb 23 09:51:52 octopi systemd[1]: mjpg-streamer.service: Main process exited, code=killed, status=15/TERM
Feb 23 09:51:52 octopi systemd[1]: mjpg-streamer.service: Succeeded.
Feb 23 10:02:46 octopi sudo[8220]:       pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/systemctl status mjpg-streamer

```

GilesBathgate, 2022-02-23:
It doesn't say why the process was killed, ` journalctl -o verbose`?

fabiosci, 2022-02-23:
```
pi@octopi:~ $ journalctl -o verbose | grep mjpg-streamer
    UNIT=mjpg-streamer.service
    UNIT=mjpg-streamer.service
    MESSAGE=mjpg-streamer.service: Unit is bound to inactive unit dev-video0.device. Stopping, too.
    UNIT=mjpg-streamer.service
    UNIT=mjpg-streamer.service
    MESSAGE=mjpg-streamer.service: Main process exited, code=killed, status=15/TERM
    UNIT=mjpg-streamer.service
    MESSAGE=mjpg-streamer.service: Succeeded.
    UNIT=mjpg-streamer.service
    MESSAGE=      pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/systemctl status mjpg-streamer
    _CMDLINE=sudo systemctl status mjpg-streamer
    _CMDLINE=sudo systemctl status mjpg-streamer
    _CMDLINE=sudo systemctl status mjpg-streamer

```

GilesBathgate, 2022-02-23:
You basically need to see the same messages as
```
MJPG Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 352 x 288
 i: Frames Per Second.: -1
 i: Format............: JPEG
 ```
 
 in your systemctl output. I don't know how you do that, but there should be a systemctl debugging tutorial online.

GilesBathgate, 2022-02-23:
also... 

>Firstly, if you run `sudo make install`

and change `/home/pi/mjpg-streamer/input_uvc.so` to `/usr/local/lib/mjpg-streamer/input_uvc.so`

fabiosci, 2022-02-23:
i executed `sudo make install` in folder /home/pi/mjpg_streamer (my `mjpg-streamer-experimental` folder is empty) as first step. i also modified the .service file to point to `/usr/local/lib/mjpg-streamer/input_uvc.so` as per your suggestion and rebooted the raspberry.

here the output you was referring to:
```
pi@octopi:~ $ /usr/local/bin/mjpg_streamer -i "/home/pi/mjpg-streamer/input_uvc.so -r 352x288"
MJPG Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 352 x 288
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: RGB3
    ... will try to handle this by checking against supported formats.
    ... Falling back to RGB24 mode (consider using -fourcc RGB24 option). Note that this requires much more CPU power
Unable to map buffer: Invalid argument
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
```

thanks a lot for your time and your help!

GilesBathgate, 2022-02-23:
@jacksonliam Looks like a similar issue: #101 
@fabiosci try `v4l2-ctl --list-formats-ext`

fabiosci, 2022-02-23:
i commented on this issue simply because your suggestions to Armageddit worked also for me (apart the service part). but i think you're right about RGB3 not supported (v4l2-ctl sais S561 and GBRG):
```
pi@octopi:~ $ export LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so
pi@octopi:~ $ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'S561' (GSPCA SPCA561, compressed)
                Size: Discrete 320x240
                Size: Discrete 352x288
        [1]: 'GBRG' (8-bit Bayer GBGB/RGRG)
                Size: Discrete 160x120
                Size: Discrete 176x144

```

GilesBathgate, 2022-02-23:
Does `/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so` exist?

GilesBathgate, 2022-02-23:
The next step would be to replace:
```
[Service]
ExecStart=/usr/local/bin/mjpg_streamer -i "/usr/local/lib/mjpg-streamer/input_uvc.so -r 352x288 -fourcc RGB24"
Environment="LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so"
```
with
[Service]
ExecStart=<PATH_TO>v4l2-ctl --list-formats-ext
Environment="LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so"
```

where <PATH_TO> is the path from eg: `which v4l2-ctl`

Then debug the service and see if you get the same output.

fabiosci, 2022-02-23:
ok, done. these the steps:
1. updated .service file as per your instructions
2. issued `sudo systemctl daemon-reload`
3. issued `sudo reboot`
4. when rapsberry went up again, manually executed `sudo systemctl start mjpg-streamer` (even if the service is enabled) but same result

```
pi@octopi:~ $ systemctl status mjpg-streamer
● mjpg-streamer.service - Streams video with Raspberry Pi Camera
   Loaded: loaded (/etc/systemd/system/mjpg-streamer.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Wed 2022-02-23 15:20:35 GMT; 2s ago
  Process: 2731 ExecStart=/usr/bin/v4l2-ctl --list-formats-ext (code=killed, signal=TERM)
 Main PID: 2731 (code=killed, signal=TERM)

Feb 23 15:20:35 octopi systemd[1]: Started Streams video with Raspberry Pi Camera.
Feb 23 15:20:35 octopi systemd[1]: mjpg-streamer.service: Unit is bound to inactive unit dev-video0.device. Stopping, too.
Feb 23 15:20:35 octopi systemd[1]: Stopping Streams video with Raspberry Pi Camera...
Feb 23 15:20:35 octopi systemd[1]: mjpg-streamer.service: Main process exited, code=killed, status=15/TERM
Feb 23 15:20:35 octopi systemd[1]: mjpg-streamer.service: Succeeded.
Feb 23 15:20:35 octopi systemd[1]: Stopped Streams video with Raspberry Pi Camera.
```

GilesBathgate, 2022-02-23:
I presumed some knowledge of systemctl, (I've mostly forgotten myself), what is needed is the output messages of the service ExecStart command. Reading up I think it's `sudo journalctl -f -u mjpg-streamer` to follow the logs of the app then in a new terminal `systemctl start mjpg-streamer`. 

`mjpg-streamer.service: Main process exited, code=killed, status=15/TERM` isn't telling us much, It might be that additional debug flag has to be added to the service description also.

GilesBathgate, 2022-02-23:
`/usr/lib/systemd/systemd --test --system --unit=mjpg-streamer` might also let you see.

fabiosci, 2022-02-24:
in one of my experiments i saw something like "failure due to dependency". not sure about its means and i wasn't able to reproduce again.
btw i cannot find an easy way - for my poor knowledge - to debug the ExecStart command. it seems it is completely ignored, along with the LD_PRELOAD (which is not set).

A simple workaround i found is to create a bash script like this:
```
LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libv4l/v4l1compat.so
/usr/local/bin/mjpg_streamer -i "/usr/local/lib/mjpg-streamer/input_uvc.so -r 352x288 -fourcc RGB24"
```
and run it with cron `@reboot`.

It works but i'm pretty sure it's not the best way...

GilesBathgate, 2022-02-25:
You can also try:
```
ExecStart=LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so /usr/local/bin/mjpg_streamer -i "/usr/local/lib/mjpg-streamer/input_uvc.so -r 352x288 -fourcc RGB24"
```

fabiosci, 2022-02-25:
after some tests i think that the problem has to do with dependencies. mjpg-streamer service doesn't start because something goes wrong with `dev-video0.device`
in fact, i removed from .service file the directives "requires" and "bindsto" and the service successfully started at boot.
this explains also why the manual execution of ExecStart command works fine.

now the point is: what is the issue with `dev-video0.device`?
I changed cam, but the behaviour so far is the same.
some outputs:
```
pi@octopi:~ $ ls /dev/video*
/dev/video0  /dev/video10  /dev/video11  /dev/video12  /dev/video13  /dev/video14  /dev/video15  /dev/video16  /dev/video18
```
```
pi@octopi:~ $ lsusb
Bus 001 Device 004: ID 0c45:60af Microdia VideoCAM Look
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
```
pi@octopi:~ $ v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16

USB camera (usb-3f980000.usb-1.2):
        /dev/video0
```
```
pi@octopi:~ $ v4l2-ctl --list-formats-ext -d 0
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'S910' (GSPCA SN9C10X, compressed)
                Size: Discrete 160x120
                Size: Discrete 320x240
                Size: Discrete 640x480
        [1]: 'BA81' (8-bit Bayer BGBG/GRGR)
                Size: Discrete 160x120
```
```
pi@octopi:~ $ fswebcam
--- Opening /dev/video0...
Trying source module v4l2...
/dev/video0 opened.
No input was specified, using the first.
Adjusting resolution from 384x288 to 160x120.
Error setting pixel format.
VIDIOC_S_FMT: Device or resource busy
```

still working on it

GilesBathgate, 2022-02-25:
@fabiosci Ok, thanks for the update, I am not sure If I can be of any further help. There are probably irc chatrooms/discord for pi/systemd/v4l you can try.

fabiosci, 2022-02-25:
i undestand and want to thank you for your precious help!

PR #249 (2020-04-29): Added support for RGB24 (RGB3)
This adds support for RGB24 aka RGB3. It's only needed by some very rare cameras.

Fixes #248