jacksonliam/mjpg-streamer #251: Segmentation error on running mjpg streamer with raspicam on debian 64 bit

numerunix, 2020-05-04:
Hello,
i made a custom build of debian 64 bit arm for raspberry pi 4, using sakaki- kernel (that can be found here on github).
I don't need x, so i compiled userland at 64 bit and then mjpg streamer at 64 bit.

raspicam works great, but if i run mjpg streamer with raspicam plugin, it crashes.

I attach compile log.

giulio@buster-rpi64:~/mjpg-streamer/mjpg-streamer-experimental/build$ make
Scanning dependencies of target mjpg_streamer
[  3%] Building C object CMakeFiles/mjpg_streamer.dir/mjpg_streamer.c.o
[  6%] Building C object CMakeFiles/mjpg_streamer.dir/utils.c.o
[  9%] Linking C executable mjpg_streamer
[  9%] Built target mjpg_streamer
Scanning dependencies of target input_file
[ 12%] Building C object plugins/input_file/CMakeFiles/input_file.dir/input_file.c.o
[ 16%] Linking C shared library input_file.so
[ 16%] Built target input_file
Scanning dependencies of target input_http
[ 19%] Building C object plugins/input_http/CMakeFiles/input_http.dir/input_http.c.o
[ 22%] Building C object plugins/input_http/CMakeFiles/input_http.dir/misc.c.o
[ 25%] Building C object plugins/input_http/CMakeFiles/input_http.dir/mjpg-proxy.c.o
[ 29%] Linking C shared library input_http.so
[ 29%] Built target input_http
Scanning dependencies of target input_opencv
[ 32%] Building CXX object plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:86:5: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
     " Help for input plugin..: "INPUT_PLUGIN_NAME"\n" \
     ^
[ 35%] Linking CXX shared library input_opencv.so
[ 35%] Built target input_opencv
Scanning dependencies of target cvfilter_cpp
[ 38%] Building CXX object plugins/input_opencv/filters/cvfilter_cpp/CMakeFiles/cvfilter_cpp.dir/filter_cpp.cpp.o
[ 41%] Linking CXX shared library cvfilter_cpp.so
[ 41%] Built target cvfilter_cpp
Scanning dependencies of target input_raspicam
[ 45%] Building C object plugins/input_raspicam/CMakeFiles/input_raspicam.dir/input_raspicam.c.o
In file included from /home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/input_raspicam.c:51:
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/RaspiCamControl.c: In function ‘raspicamcontrol_set_saturation’:
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/RaspiCamControl.c:762:32: warning: implicit declaration of function ‘mmal_port_parameter_set_rational’; did you mean ‘mmal_port_parameter_set’? [-Wimplicit-function-declaration]
       ret = mmal_status_to_int(mmal_port_parameter_set_rational(camera->control, MMAL_PARAMETER_SATURATION, value));
                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                mmal_port_parameter_set
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/RaspiCamControl.c: In function ‘raspicamcontrol_set_ISO’:
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/RaspiCamControl.c:864:30: warning: implicit declaration of function ‘mmal_port_parameter_set_uint32’; did you mean ‘mmal_port_parameter_set’? [-Wimplicit-function-declaration]
    return mmal_status_to_int(mmal_port_parameter_set_uint32(camera->control, MMAL_PARAMETER_ISO, ISO));
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                              mmal_port_parameter_set
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/RaspiCamControl.c: In function ‘raspicamcontrol_set_video_stabilisation’:
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/RaspiCamControl.c:899:30: warning: implicit declaration of function ‘mmal_port_parameter_set_boolean’; did you mean ‘mmal_port_parameter_set’? [-Wimplicit-function-declaration]
    return mmal_status_to_int(mmal_port_parameter_set_boolean(camera->control, MMAL_PARAMETER_VIDEO_STABILISATION, vstabilisation));
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                              mmal_port_parameter_set
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/RaspiCamControl.c: In function ‘raspicamcontrol_set_exposure_compensation’:
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/RaspiCamControl.c:913:30: warning: implicit declaration of function ‘mmal_port_parameter_set_int32’; did you mean ‘mmal_port_parameter_set’? [-Wimplicit-function-declaration]
    return mmal_status_to_int(mmal_port_parameter_set_int32(camera->control, MMAL_PARAMETER_EXPOSURE_COMP , exp_comp));
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                              mmal_port_parameter_set
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/input_raspicam.c: In function ‘worker_thread’:
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/input_raspicam.c:885:10: warning: implicit declaration of function ‘mmal_port_pool_create’; did you mean ‘mmal_pool_create’? [-Wimplicit-function-declaration]
   pool = mmal_port_pool_create(encoder_output, encoder_output->buffer_num, encoder_output->buffer_size);
          ^~~~~~~~~~~~~~~~~~~~~
          mmal_pool_create
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/input_raspicam.c:885:8: warning: assignment to ‘MMAL_POOL_T *’ {aka ‘struct MMAL_POOL_T *’} from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
   pool = mmal_port_pool_create(encoder_output, encoder_output->buffer_num, encoder_output->buffer_size);
        ^
/home/giulio/mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam/input_raspicam.c:1057:5: warning: implicit declaration of function ‘mmal_port_pool_destroy’; did you mean ‘mmal_pool_destroy’? [-Wimplicit-function-declaration]
     mmal_port_pool_destroy(encoder->output[0], pool);
     ^~~~~~~~~~~~~~~~~~~~~~
     mmal_pool_destroy
[... 96 lines cut ...]
-- Installing: /usr/local/lib/mjpg-streamer/output_http.so
-- Installing: /usr/local/lib/mjpg-streamer/output_rtsp.so
-- Installing: /usr/local/lib/mjpg-streamer/output_udp.so
root@buster-rpi64:/home/giulio/mjpg-streamer/mjpg-streamer-experimental/build# nano /lib/systemd/system/mjpg-streamer.service 
root@buster-rpi64:/home/giulio/mjpg-streamer/mjpg-streamer-experimental/build# mjpg_streamer -i '/usr/local/lib/mjpg-streamer/input_raspicam.so' -o '/usr/local/lib/mjpg-streamer/output_http.so -w /usr/local/share/mjpg-streamer/www -p 8080'
MJPG Streamer Version: git rev: 5a6e0a2db163e6ae9461552b59079870d0959340
 i: fps.............: 5
 i: resolution........: 640 x 480
 i: camera parameters..............:

Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
 o: www-folder-path......: /usr/local/share/mjpg-streamer/www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
 i: Starting Camera
Encoder Buffer Size 81920
Errore di segmentazione
root@buster-rpi64:/home/giulio/mjpg-streamer/mjpg-streamer-experimental/build#

xamindar, 2020-07-05:
I can't even get it to compile input_raspicam on 64bit raspbian. How did you do it? I assume some dependency is missing but the log doesn't even seem to attempt to compile it.

xlla, 2020-09-21:
same issue here, pi4@aarch64
<pre>
mjpg_streamer -i "input_raspicam.so -x 320 -y 240 " -o "output_http.so -p 8080"
MJPG Streamer Version: git rev: 5a6e0a2db163e6ae9461552b59079870d0959340
 i: fps.............: 5
 i: resolution........: 320 x 240
 i: camera parameters..............:

Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
 o: www-folder-path......: disabled
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
 i: Starting Camera
Encoder Buffer Size 81920
Segmentation fault

</pre>

jacksonliam, 2020-12-18:
The MMAL API is not supported on 64 bit RPI images yet. input_raspicam needs MMAL. I believe input_uvc is working with the rpi cam v4l2 driver. Continued discussion at #259

closed 2020-12-18 by jacksonliam