jacksonliam/mjpg-streamer #252: sometimes huge lag ans slow frame rate

rdmueller, 2020-06-02:
mjpg-streamer is easy to setup - thanx for this project!

However, I often experience huge lag (two and more seconds) together with slow frame rates.
And sometimes it just works great. 
I wonder what the reson for this could be. I run mjpg-streamer on a pi-zero without any other software, connected through network over usb.

Any ideas are highly welcome!

Plasmarobo, 2020-10-09:
Just experienced this randomly. Checked `top` and I'm only using around 3% CPU for `mjpg_stream` and about another ~30-40% on other tasks. 

I do have octoprint running as well, doing a print task. I'm getting a framerate between 0.5 and 1 fps. Seems arbitrary when this problem strikes.
Running on Raspberry Pi 3B

jacksonliam, 2020-12-18:
Sorry I'm not really sure what could be causing it and without being able to reproduce this I can't look into it futher. If anyone does any deeper investigation or more people are seeing this then feel free to add it here and I can repoen the issue.

closed 2020-12-18 by jacksonliam