jacksonliam/mjpg-streamer #256: Logitech C270 no longer seems to work.  Is it still supported?

bwims, 2020-06-17:
I recently upgraded Octopi from version 0.14 which is pre-fork of mjpg-streamer
to 0.17 which is the latest version and is presumably post fork.

Pre-fork works perfectly with C270.  Now it produces a scrambled picture as attached.

I also have a Logitech C510 . I plugged that in and rebooted and it produced a perfect picture.

I then used a different C270 with the same garbled result.

Can you give me any clue as to what is going on here?

Thanks.

Brian Williams
[image]

ChaosEngine, 2020-06-17:
I also sometime have this issue on Logitech C270. Usually after fresh syste mreboot. For me it helps to restart mjpg-streamer and/or issue this on system startup (some inner script):

```
#set default video camera params
if [ -x "/usr/bin/v4l2-ctl" ]
then
        /usr/bin/v4l2-ctl --set-fmt-video=width=1280,height=960 -c exposure_auto_priority=1;
   #echo 'done';
fi

```
I would recommend with tinkering with various `v4l2-ctl` parameters

jacksonliam, 2020-07-03:
Does the console output of mjpg-streamer differ when its wrong? 

Assuming this is not an mjpg-streamer issue but a wider OS/driver issue currently, so will close for now.

closed 2020-07-03 by jacksonliam

AnVa, 2021-08-24:
https://github.com/jacksonliam/mjpg-streamer/issues/316