jacksonliam/mjpg-streamer #257: linker error multiple definition of `client_infos'

arimakai, 2020-06-29:
Hi,
I found error
/bin/ld: CMakeFiles/output_http.dir/output_http.c.o:(.bss+0xa50): multiple definition of `client_infos'; CMakeFiles/output_http.dir/httpd.c.o:(.bss+0x8): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [plugins/output_http/CMakeFiles/output_http.dir/build.make:116: plugins/output_http/output_http.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:479: plugins/output_http/CMakeFiles/output_http.dir/all] Error 2
make: *** [Makefile:147: all] Error 2

When I checked code output_http.c and httpd.c both scripts have include httpd.h
I fixed this problem by move
struct {
    client_info **infos;
    unsigned int client_count;
    pthread_mutex_t mutex;
} client_infos;
in httpd.h to httpd.c

Can you review this point?

*Note: compiled on fedora 32 aarch64, GCC 10.0.1, cmake 3.17.0

jacksonliam, 2020-07-03:
Could you try editing the file https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/CMakeLists.txt#L86, above the line
`set (CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -Wl,--enable-new-dtags")`
Add a new line with:
`set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS} -fcommon")`

ollpu, 2020-07-10:
@jacksonliam Works with that change (after cleaning out CMakeCache.txt).

The issue is with `client_info` and `client_infos` being defined in the header file.

mikegleasonjr, 2020-11-26:
Have the same issue after having `ENABLE_HTTP_MANAGEMENT` enabled.

The workaround did not fix it for me though...

stefan-sherwood, 2022-02-27:
I had the same problem with the `output_http `plugin and moving the `client_infos` struct from `httpd.h `to `httpd.c` also resolved it.

LMBernardo, 2022-09-04:
Building with `ENABLE_HTTP_MANAGEMENT` is fixed in [my fork](https://github.com/LMBernardo/mjpg-streamer).