jacksonliam/mjpg-streamer #259: Not getting input_raspicam when compiling on 64bit raspbian

xamindar, 2020-07-05:
This doesn't seem to be compiling everything when I try on 64 but Raspbian. I am not seeing any errors either.

pi@raspberrypi4:~/mjpg-streamer/mjpg-streamer-experimental $ make
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
make -C _build
make[1]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[2]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[  4%] Building C object CMakeFiles/mjpg_streamer.dir/mjpg_streamer.c.o
[  8%] Building C object CMakeFiles/mjpg_streamer.dir/utils.c.o
[ 13%] Linking C executable mjpg_streamer
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 13%] Built target mjpg_streamer
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 17%] Building C object plugins/input_file/CMakeFiles/input_file.dir/input_file.c.o
[ 21%] Linking C shared library input_file.so
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 21%] Built target input_file
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 26%] Building C object plugins/input_http/CMakeFiles/input_http.dir/input_http.c.o
[ 30%] Building C object plugins/input_http/CMakeFiles/input_http.dir/misc.c.o
[ 34%] Building C object plugins/input_http/CMakeFiles/input_http.dir/mjpg-proxy.c.o
[ 39%] Linking C shared library input_http.so
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 39%] Built target input_http
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 43%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/dynctrl.c.o
[ 47%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/input_uvc.c.o
[ 52%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/jpeg_utils.c.o
[ 56%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/v4l2uvc.c.o
[ 60%] Linking C shared library input_uvc.so
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 60%] Built target input_uvc
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 65%] Building C object plugins/output_file/CMakeFiles/output_file.dir/output_file.c.o
[ 69%] Linking C shared library output_file.so
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 69%] Built target output_file
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 73%] Building C object plugins/output_http/CMakeFiles/output_http.dir/httpd.c.o
[ 78%] Building C object plugins/output_http/CMakeFiles/output_http.dir/output_http.c.o
[ 82%] Linking C shared library output_http.so
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 82%] Built target output_http
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 86%] Building C object plugins/output_rtsp/CMakeFiles/output_rtsp.dir/output_rtsp.c.o
[ 91%] Linking C shared library output_rtsp.so
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 91%] Built target output_rtsp
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 95%] Building C object plugins/output_udp/CMakeFiles/output_udp.dir/output_udp.c.o
[100%] Linking C shared library output_udp.so
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[100%] Built target output_udp
make[2]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[1]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
pi@raspberrypi4:~/mjpg-streamer/mjpg-streamer-experimental $

jacksonliam, 2020-07-06:
Have you tried running `make distclean` first?

xamindar, 2020-07-07:
Yes I have, but it doesn't work. "make clean" works:

pi@raspberrypi4:~/mjpg-streamer/mjpg-streamer-experimental $ make distclean
make: *** No rule to make target 'distclean'.  Stop.

xamindar, 2020-07-07:
I also pulled down a fresh repo and still does not build that file:
Is it possible any of the following are not really optional as it says?

-- The following OPTIONAL packages have not been found:

 * OpenCV
 * Gphoto2
 * SDL

xamindar, 2020-07-07:
OK, it looks like this package looks for includes in a non-standard location. I got it to build by working around with a symlink:
ln -s /usr/include /opt/vc/

jacksonliam, 2020-07-08:
Thanks that's useful info.

It might be that the headers we have had to include in this repo to build against mmal need updating for 64 bit. 

I assume the other, non input_raspicam modules work? You can test out input_uvc wih a raspi cam by running 

`modprobe bcm2835-v4l`

xamindar, 2020-07-08:
I posted about it in the pi forums and someone said mmal is not yet working on 64bit raspbian. So I guess that's that.

I wasn't aware you could use input_uvc, thanks. However, I was unable to get that to work. When I try, i just get a constant scroll of "Ignoring empty buffer ..." going down the screen.

Looking in to it further, I was able to successfully capture an image using the v4l2-ctl command. So I know the camera does work.

jacksonliam, 2020-07-08:
OK thanks for the info.
Sounds like the part of bcm2835-v4l we need isnt working either, if it's just delivering empty buffers. Even if v4l2-ctl is able to create an image.

gulbrillo, 2020-07-19:
Hey :) I'm running into the same issues. So far, getting the latest and greatest Pi was a big mistake. ^^
I edited `plugins/input_raspicam/CMakeLists.txt` to make it compile. But it segmentation faults. (mmal issue?)
Do we have to wait until the 64 bit framework is available? or did someone find a workaround? Thanks all!! :)

guysoft, 2020-12-17:
Any news on this?

jacksonliam, 2020-12-18:
@guysoft I can't really do anything about input-raspicam until MMAL is supported on the 64-bit OS which I don't think it is yet? 

I believe mjpg-streamer build correctly detects this and does not build the plugin. So no errors or failures to build are seen as far as I know?

V4L2 using the input-uvc plugin and the raspberry pi cam v4l2 driver should still work IIRC? I believe it is out of the box on your 0.18 RC1.

Cheers!

esmaeilpour, 2021-01-13:
Came across this and guess might be helpful
https://github.com/jacksonliam/mjpg-streamer/issues/203#issuecomment-484513852

guysoft, 2021-01-18:
According to @esmaeilpour simply installing  libraspberrypi-dev package should make it compile.

jacksonliam, 2021-01-18:
> According to @esmaeilpour simply installing libraspberrypi-dev package should make it compile.

Probably pulling in the headers will compile the plugin but I believe it still won't work at runtime on 64 bit.

FredLackeyOfficial, 2021-01-20:
> libraspberrypi-dev

Did not help.

nymda, 2022-01-11:
any update on this? im also not getting input_raspicam

mcr222, 2022-01-28:
just want to leave a note here in case anybody ends up here. Even for 32 bit it does not build input_raspicam.so because of the moving of the libraries.

The easiest way to solve this is by just modifying the CMakeLists.txt from mjpg-streamer/mjpg-streamer-experimental/plugins/input_raspicam and change all paths "opt/vc" for "usr".

alephO, 2022-02-19:
> just want to leave a note here in case anybody ends up here. Even for 32 bit it does not build input_raspicam.so because of the moving of the libraries.
>
> The easiest way to solve this is by just [...]

I am on a 32-bit OS, and create symbolic link with `sudo ln -s /usr /opt/vc` following by a `make distclean` works for me. Seems that somehow the include path has changed.

stevesdawg, 2022-05-22:
> I am on a 32-bit OS, and creating a symbolic link with sudo ln -s /usr /opt/vc following by a make distclean works for me. Seems that somehow the include path has changed.

I'm getting compiler warnings when I compile the `input_raspicam` plugin after making this symlink, however the raspicam plugin seems to be running normally! Just for context/knowledge, why is there a discrepancy in the headers between what's expected, and what is installed on Raspbian?

Mactarvish, 2022-09-28:
`sudo apt-get install libraspberrypi-dev`
That works!

Note that I'm on 4B+ with `Linux ubuntu 5.4.0-1069-raspi #79-Ubuntu SMP PREEMPT Thu Aug 18 18:15:22 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux` instead of original rasberry system, current system can be downloaded in official website.