jacksonliam/mjpg-streamer #260: Unplug v4l device cannot finished mjpg-streamer process (freezed)

asmyasnikov, 2020-07-07:
https://github.com/jacksonliam/mjpg-streamer/blob/925337f541ddbb061ade0bee5215299f7834d8ea/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c#L514

I have once in a way some issue behaviour - if device unplugged, then last message:
`Unable to stop capture: No such device` (see screenshot)
And mjpg-streamer process freezed (no exit). Because of this I can't restart mjpg-streamer

[image]

asmyasnikov, 2020-07-09:
I re-build in DEBUG mode
Last logs of process:
```
...
 DBG(/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 528): sending frame
 DBG(/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 531): sending boundary
 DBG(/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 699): select() = 2
 DBG(/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 720): Grabbing a frame...
Unable to dequeue buffer: No such device
 i: Error grabbing frames
 DBG(/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 831): leaving input thread, calling cleanup function now
 i: cleaning up resources allocated by input thread
 DBG(/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, video_disable(), 510): STopping capture
Unable to stop capture: No such device
```

jacksonliam, 2020-07-09:
mjpg-streamer has multiple plugins and can run multiple inputs and outputs at once. It is not designed to quit just because one of the plugins has an error.

Why are you not able to restart mjpg-streamer when it this happens? It should still shut down in the normal way after this plugin has stopped.

closed 2020-07-09 by jacksonliam

asmyasnikov, 2020-07-09:
I'm use sinle input plugin (input_uvc) and I expect shut down of mjpg_streamer.
Handle restart of mjpg_streamer does not suit me for production devices like as RaspberryPi without any display and monitoring tools
Please, add auto shut down of mjpg_streamer if single (or last of few inputs) are stopped

ChaosEngine, 2020-07-09:
@asmyasnikov can't you monitor for such occasions (disconnecting of device) and react upon them with mjpg-streamer restar?

Found something sounding quite reasonable here: https://stackoverflow.com/a/24141366/4429828