jacksonliam/mjpg-streamer #261: HTTPS delivery

wouldlovetogetittowork, 2020-07-22:
MJPG-streamer works perfectly on raspberry pi. Thank you. A recent pi project requires video delivery within a secure web-page (HTTPS). As streaming HTTP downgraded host page status I ported HTTP output through a proxy to provide HTTPS. On first try an otherwise working stream failed. However JavaScript snap_shot example worked, but ran intermittently.
https://github.com/jacksonliam/mjpg-streamer/issues/100 was insightful yet suggestions are implemented in the version I'm using.
Debugging shows that the input_uvc handler runs without interruption. On https web-page load, mjpg-streamer server_thread connects without a hitch. Client_thread delivers secure content back to the webpage via proxy. Image delivery ranges from 20 to 2000 iterations before abruptly stopping. A click on the webpage image (as per mjpg-streamer sample) or reloading the web-page and image delivery resumes. 
Spliced DBG output at point of failure: [sample.txt](https://github.com/jacksonliam/mjpg-streamer/files/4961461/sample.txt)
Would this constitute an issue, or is my specific HTTPS implementation beyond the scope of intention for mjpg-streamer? Any thoughts would be appreciated.

jacksonliam, 2020-07-22:
Hey sorry no idea what this might be, something to do with #51 possiblity?

wouldlovetogetittowork, 2020-07-22:
Wow, quick response. Thank you. Well spotted on https://github.com/jacksonliam/mjpg-streamer/issues/51. Will dabble a little further.

wouldlovetogetittowork, 2020-07-23:
If permissible I would like to reopen this issue? Although likely related to https://github.com/jacksonliam/mjpg-streamer/issues/51, I have been unable to conclusively prove or disprove a correlation at this stage.
To recap: using HTTP, mjpg-streamer works perfectly, even with the odd frame dropped here and there and the odd HTTP delivery getting mangled. Switching to HTTPS, transfer of snapshots is unable to recover from a fall. Where the correlation seems to be supported is in time frame dilation. Secure is just wider. It might be that an already tenuous HTTP delivery is burdened even further by use of the secure HTTPS protocol? Possibly the solution lies in reviewing the nuts and bolts of network layer protocols, however it may be more expedient for me to simply provide a HTTPS delivery timeout and focus on recovery? Any further comments would be most welcome.

wouldlovetogetittowork, 2020-07-24:
On closer investigation of the last HTTP packet delivered to the web-page before delivery fails, it's worth noting that the HTTP header delivered completely in tact. However the JPG image was corrupted in a specific manner. With this image, the JPG header is correct and in situ, a body exists as does a footer, however immediately following the header there is a 2048 byte block of pseudo data.... to illustrate:
{  JPG header
2kB byte block of… 
0x08, 0x10, 0x08, 0x81, 0x08, 0x10, 0x08, 0x80, 
0x08, 0x10, 0x08, 0x80, 0x08, 0x10, 0x08, 0x81, 
…
…
JPG image data
..
..
JPG footer }
Interesting? Exactly 2048 bytes. Always same position. I thought this 2kB of data may have been introduced in transmission, so I added a field to the header which specified the 'Originating content length: ---- kB'. That way I could compare the originating length to the 'Content-Length' delivered to the web-page.  The 2 values matched, indicating that the image send by send_snapshot( ) is unaltered in transit, yet corrupted at source.

cms-codes, 2020-07-24:
What kind of proxy are you using?

After attempting something similar with nginx a while ago, I gave up for
similar reasons.

If the receiving client is mangling the HTTP packet, then it's not
mjpg-streamer that's to blame, is it?

jacksonliam, 2020-07-24:
Interesting investigation. I'm wondering if the proxy isn't really the issue and it's the client that stops requesting frames?

Have you debugged the client? Even just checking the browsers JavaScript console?
I wonder if something is causing the browser / JavaScript to stop requesting frames (e.g a corrupted jpeg, as you said, or a HTTPS error).

Assuming you're using jpeg mode (not -y or -fourcc) then a corrupted jpeg is most likely caused by the camera, or the driver as we are literally just copying a buffer, there's not much to go wrong. You could try also enabling the file output plugin, to see if those jpegs look corrupted in the files.

We have some stuff to skip frames below a certain size, because some cameras do send the odd tiny broken frame. So it wouldn't surprise me if your camera sends the occasional corrupted frame.

You could try other resolutions, frame rates, or -y yuv mode, just to see if things are better behaved liked that.

wouldlovetogetittowork, 2020-07-24:
@cms-codes - I concur. I'm using lighttpd. Stopped the lighttpd service though after your comment, and pointed mjpg-streamer directly to port, yet corruption persists. Possibly what I considered working 'perfectly' before the proxy intervention, was a combination of eager assumption and impatience.
@jacksonliam - Good questions and great suggestions. I regret to have wasted your and cms-codes' time. It certainly appears that the corruption is at the source. My curiosity lies in the fact that the corruption is precise. Always just after the JPEG header and always 2048 bytes, but you guys have been great, yet that curiosity only goes to support your suspicion. I am happy to close this issue, as mjpg-streamer is working perfectly.

closed 2020-07-24 by wouldlovetogetittowork