jacksonliam/mjpg-streamer #264: v4l2loopback issue "Unable to query buffer"

mikulik86, 2020-08-02:
Ihave a Raspberry Pi 4 B + the Raspberry Pi High Quality Camera. 
I need to use v4l2loopback to copy videofeed from the physical Camera (/dev/video0) to dummy device (/dev/video1) in order to access it by the mjpg-streamer and the v4l2rtspserver at the same time.

I load the v4l2loopback module:

`sudo modprobe v4l2loopback`

Then copy the video from /dev/video0 to /dev/video1:

`ffmpeg -f video4linux2 -input_format mjpeg -i /dev/video0 -vcodec copy -f v4l2 /dev/video1`

Launch the mjpg-streamer and get following error:

````
./mjpg_streamer -i "./input_uvc.so -d /dev/video1" -o "./output_http.so"
MJPG Streamer Version: git rev: 85f89a8c321e799fabb1693c5d133f3fb48ee748
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
Unable to query buffer: Invalid argument
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
````

Launching the mjpg-streamer on /dev/video0 works flawlessly.

I spent hours trying to find solution. Please help.

jacksonliam, 2020-08-02:
Looks like a v4l2loopback issue to me. 
Could ask whoever maintains it why this buffer query fails? 

https://github.com/jacksonliam/mjpg-streamer/blob/925337f541ddbb061ade0bee5215299f7834d8ea/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c#L447

closed 2020-08-02 by jacksonliam

mikulik86, 2020-08-02:
Really? Even though for example v4l2rtspserver works with this dummy video device?