jacksonliam/mjpg-streamer #267: Change input port

carcamerarig, 2020-10-02:
Is it possible to change the input port to the camera port and use raspistill instead of the video port and raspivid? 

I need streamer to use something like (raspistill= -usestills -1fps) so that I get full hi res singe frame images. Don't need processing of the images if that's a problem, just stored for download would be fine.

jacksonliam, 2020-10-02:
Yes -usestills is already an option for the rpi plugin, max FPS is about 8.

closed 2020-10-02 by jacksonliam

carcamerarig, 2020-10-02:
Appreciate the link, i have followed that with interest and have tried many times to put it into still mode but i have never been able to do it, i always get errors in octoprint and cant get past it...
This is how i have to have it setup now to work outside of 640x480 resolution on a Pi4b8Gb as the [camera_raspi-options" -1280x720 -f15" does not work, infact USB is all that works and i have to mirror the setting in webcamd also otherwise it will not work at all (a bug i believe into tricking it to think its a USB cam??)

[image]

I have tried many many times to use the parameters as they should be in octopi.txt:

camera"raspi"
camera_raspi-options" -usestills -f1" but always get the octoprint errors that wont allow to start a print

[image]

I know you closed this but if you could assist me i would very much appreciate it.

cheers J

jacksonliam, 2020-10-02:
As good as octopi is (I use it personally) I don't know how any of that config file stuff or that web error works, it's nothing to do with mjpg-streamer. You'd have to go to the octopi community for help with it.

If you run mjpg-streamer from the command line, and look at the stream in the browser, and tell me what command you run and what errors you get, I might be able to help you out.

carcamerarig, 2020-10-04:
> As good as octopi is (I use it personally) I don't know how any of that config file stuff or that web error works, it's nothing to do with mjpg-streamer. You'd have to go to the octopi community for [...]

So i have tried to use the mjpg-streamer-experimental fork also on a fresh install of octopi and am just plain struggling, i did put it to the octopi and octoprint communities and they just said no its an mjpg streamer issue and that they just use the streamer as is. All im really after is to be able to use the -usestills service i dont care for anything extra as long as i can download the stills. Any way you can help me along with it?

jacksonliam, 2020-10-04:
I'm guessing you need to disable octoprint starting mjpg-streamer and ssh into the pi and start it yourself, with the command line you want.

carcamerarig, 2020-10-04:
Is it possible to have the pi camera run on the external camera configuration? that takes a still image and stores it in the snapshot folder right?

jacksonliam, 2020-10-04:
I'm not sure what you mean by the "external camera configuration". mjpg-streamer has no concept of this.

mjpg-streamer also has no notion of a snapshot folder, I'm guessing that's an octoprint thing.

We have the output file plugin which can save jpegs to a specified folder with a specified delay, if that's what you're looking for. You can run multiple output plugins so can run this alongside the output HTTP plugin for viewing in the browser.

https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/output_file/output_file.c#L65

carcamerarig, 2020-10-04:
Actually it is an octolapse thing but as a question do you know anything about it?

https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera