jacksonliam/mjpg-streamer #271: input_uvc.so: undefined symbol: parse_resolution_opt

l4es, 2020-10-26:
Hi, 

I built mjpg-streamer package within Buildroot and the mjpg-streamer.mk is as below :
 `################################################################################
#
# mjpg-streamer
#
################################################################################

MJPG_STREAMER_VERSION = ddb69b7b4f114f3c2ca01adf55712792ca8aed43
MJPG_STREAMER_SITE = $(call github,jacksonliam,mjpg-streamer,$(MJPG_STREAMER_VERSION))
MJPG_STREAMER_SUBDIR = mjpg-streamer-experimental
MJPG_STREAMER_LICENSE = GPL-2.0+
MJPG_STREAMER_LICENSE_FILES = $(MJPG_STREAMER_SUBDIR)/LICENSE
MJPG_STREAMER_DEPENDENCIES = jpeg

ifeq ($(BR2_PACKAGE_LIBGPHOTO2),y)
MJPG_STREAMER_CONF_OPTS += -DPLUGIN_INPUT_PTP2=ON
MJPG_STREAMER_DEPENDENCIES += host-pkgconf libgphoto2
else
MJPG_STREAMER_CONF_OPTS += -DPLUGIN_INPUT_PTP2=OFF
endif

ifeq ($(BR2_PACKAGE_LIBV4L),y)
MJPG_STREAMER_DEPENDENCIES += libv4l
endif

ifeq ($(BR2_PACKAGE_OPENCV3_LIB_HIGHGUI)$(BR2_PACKAGE_OPENCV3_LIB_IMGPROC)$(BR2_PACKAGE_OPENCV3_LIB_VIDEOIO),yyy)
MJPG_STREAMER_CONF_OPTS += \
        -DPLUGIN_INPUT_OPENCV=ON \
        -DPLUGIN_CVFILTER_CPP=OFF \
        -DPLUGIN_CVFILTER_PY=OFF

MJPG_STREAMER_DEPENDENCIES += host-pkgconf opencv3
else
MJPG_STREAMER_CONF_OPTS += -DPLUGIN_INPUT_OPENCV=OFF
endif

ifeq ($(BR2_PACKAGE_PROTOBUF_C)$(BR2_PACKAGE_ZEROMQ),yy)
MJPG_STREAMER_CONF_OPTS += -DPLUGIN_OUTPUT_ZMQSERVER=ON
MJPG_STREAMER_DEPENDENCIES += host-pkgconf protobuf-c zeromq
else
MJPG_STREAMER_CONF_OPTS += -DPLUGIN_OUTPUT_ZMQSERVER=OFF
endif

ifeq ($(BR2_PACKAGE_SDL),y)
MJPG_STREAMER_CONF_OPTS += -DPLUGIN_OUTPUT_VIEWER=ON
MJPG_STREAMER_DEPENDENCIES += host-pkgconf sdl
else
MJPG_STREAMER_CONF_OPTS += -DPLUGIN_OUTPUT_VIEWER=OFF
endif

$(eval $(cmake-package))
`

Howver, when lauching the command on target, I got following error :
`# mjpg_streamer -i "/usr/lib/mjpg-streamer/input_uvc.so -d /dev/video0 -y  -r 16
0x120 -f 5 -q 30" -o "/usr/lib/mjpg-streamer/output_http.so -p 8080 -w /usr/shar
e/mjpg-streamer/www"
MJPG Streamer Version: git rev: 106c739459735ae202c5f4cf6d382a34bfe6ac00
mjpg_streamer: symbol lookup error: /usr/lib/mjpg-streamer/input_uvc.so: undefined symbol: parse_resolution_opt`

It seems that the https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/utils.c is not built since it contains the definition of above symbol.

Do you have any idea about the issue, please ?
K.

jacksonliam, 2020-12-18:
Sorry I cannot provide support for other build systems. The instructions for building are in the first readme... you just run make.

https://github.com/jacksonliam/mjpg-streamer/blob/master/README.md

closed 2020-12-18 by jacksonliam

simonchen007, 2023-03-18:
@l4es  have you resolved this issue ?