jacksonliam/mjpg-streamer #275: Rock64: video capture not supported?

eightheads, 2020-11-05:
Hi all, I'm setting up a fresh install of Armbian Debian, Octoprint, and mjpg-streamer on my Rock64 but cannot seem to get the video feed working. I am currently trying a generic USB camera that works fine with my Raspberry Pi installations, but have also tried a Logitech C270 with the same results. Any help or suggestions would be appreciated, thanks!

```
./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so -w ./www"

MJPG Streamer Version: git rev: 85f89a8c321e799fabb1693c5d133f3fb48ee748
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
Error opening device /dev/video0: video capture not supported.
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
```

```
lsusb

Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 003: ID 0c45:6366 Microdia 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

jacksonliam, 2020-12-18:
Can you post the output of `v4l2-ctl --list-formats-ext` on the Rock64? Its probably due to v4l2 or the webcam driver on the version of Linux on the Rock64, unlikely anything to do with mjpg-streamer

closed 2021-02-20 by jacksonliam

eightheads, 2021-02-21:
> Can you post the output of `v4l2-ctl --list-formats-ext` on the Rock64? Its probably due to v4l2 or the webcam driver on the version of Linux on the Rock64, unlikely anything to do with mjpg-streamer

Sorry about not replying earlier:

root@rock64:~# lsusb
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@rock64:~# v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture Multiplanar

	[0]: 'NV12' (Y/CbCr 4:2:0)

jacksonliam, 2021-02-22:
Hmmm yeah something wrong with that, if you run it on a working installation you will get YUV or JPEG modes.

So I expect the issue here is v4l2 or the camera driver on the RockPis flavour of Linux. If they provide other distros (e.g. Ubuntu) might be worth giving those a go.