jacksonliam/mjpg-streamer #276: Turn http image snapshots into an MJPEG stream

migalenkom, 2020-11-11:
I have a url: 'http://192.168.1.51/cgi-bin/snapshot.cgi?chn=0&u=admin&p=pass' - it returns one single image. I would like to turn it into an MJPEG stream.

I am trying this:
`mjpg_streamer -i '/usr/local/lib/mjpg-streamer/input_http.so "http://admin:pass@192.168.1.51/cgi-bin/snapshot.cgi?chn=0&u=admin&p=pass"' -o '/usr/local/lib/mjpg-streamer/output_http.so -p 8085 -w /usr/local/share/mjpg-streamer/www'`

`MJPG Streamer Version: git rev: 85f89a8c321e799fabb1693c5d133f3fb48ee748
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/input_http/input_http.c, input_init(), 78): argv[0]=HTTP Input plugin
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/input_http/input_http.c, input_init(), 78): argv[1]="http://admin:pass@192.168.1.51/cgi-bin/snapshot.cgi?chn=0&u=admin&p=pass"
 i: host.............: localhost
 i: port.............: 8080
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/output_http.c, output_init(), 87): output #00
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/output_http.c, output_init(), 98): argv[0]=HTTP output plugin
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/output_http.c, output_init(), 98): argv[1]=-p
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/output_http.c, output_init(), 98): argv[2]=8085
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/output_http.c, output_init(), 98): argv[3]=-w
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/output_http.c, output_init(), 98): argv[4]=/usr/local/share/mjpg-streamer/www
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/output_http.c, output_init(), 144): case 2,3
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/output_http.c, output_init(), 165): case 8,9
 o: www-folder-path......: /usr/local/share/mjpg-streamer/www/
 o: HTTP TCP port........: 8085
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer.c, main(), 401): starting 1 input plugin
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer.c, main(), 411): starting 1 output plugin(s)
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/output_http.c, output_run(), 225): launching server thread #00
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/output_http/httpd.c, server_thread(), 1555): waiting for clients to connect
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/input_http/mjpg-proxy.c, connect_and_stream(), 210): socket value is 5
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/input_http/mjpg-proxy.c, connect_and_stream(), 210): socket value is 5
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/input_http/mjpg-proxy.c, connect_and_stream(), 212): connected to host
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/input_http/mjpg-proxy.c, connect_and_stream(), 230): Closing socket
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/input_http/mjpg-proxy.c, connect_and_stream(), 210): socket value is 5
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/input_http/mjpg-proxy.c, connect_and_stream(), 210): socket value is 5
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/input_http/mjpg-proxy.c, connect_and_stream(), 212): connected to host
 DBG(/home/deployer/mjpg-streamer/mjpg-streamer-experimental/plugins/input_http/mjpg-proxy.c, connect_and_stream(), 230): Closing socket
`

rgaufman, 2020-11-11:
I have the same issue, anyone have a solution or a workaround?

jacksonliam, 2020-11-11:
Ok but what is happening? Is mjpg-streamer exiting at this point, Just not getting a stream? Etc?

migalenkom, 2020-11-11:
mjpg-streamer keep trying to `connect_and_stream()` and then Closing socket like the last 10 lines in the log above but no video is streaming.
I was trying to connect with vlc `http://localhost:8085/?action=stream` and watch via browser as well.
VLC says `Your input can't be opened` and blank box in the browser window where should be a stream. 
Any ideas?

jacksonliam, 2020-11-12:
Ok the input_http plugin currently doesn't support anything other than talking to another instance of mjpg-streamer.

The URL is hard coded. See: https://github.com/jacksonliam/mjpg-streamer/blob/85f89a8c321e799fabb1693c5d133f3fb48ee748/mjpg-streamer-experimental/plugins/input_http/mjpg-proxy.c#L113

I didn't write this, and it might be useful to make it be able to grab frames from a given web address, support https, etc. But it's not something I'm going to do right now, PRs welcome!