jacksonliam/mjpg-streamer #282: Unable to make OpenCV plugin (input_opencv.so) during compile.

ansarid, 2020-12-15:
I've been unable to satisfy the cmake requirements needed to compile input_opencv.so during the `make` and `make install` process. Unsure if I'm missing something simple or if there is a legitimate issue that needs to be tackled. Either way I've spent a lot of time trying to make this work and been unsuccessful.

## Steps to Reproduce:

Platform: Raspberry Pi 3 Model B+
OS: Raspberry Pi OS Lite

1. Download and Flash [Raspberry Pi OS Lite](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-12-04/2020-12-02-raspios-buster-armhf-lite.zip) to a microSD card. Setup WiFi, SSH, etc.

2. Update and Upgrade: 
```
sudo apt update 
sudo apt upgrade -y
sudo apt dist-upgrade -y
```

3. Install Dependencies: 
```
sudo apt install git gcc g++ cmake libjpeg8-dev  -y
sudo apt install python3-opencv libopencv-dev libprotobuf-c-dev libsdl-dev -y
```

4. Clone mjpg-streamer: 
```
git clone https://github.com/jacksonliam/mjpg-streamer
```

5. Compile:
```
cd mjpg-streamer/mjpg-streamer-experimental
export OpenCV_DIR=/usr/share/OpenCV
make
sudo make install
```
## Output:
```
pi@raspberrypi:~/mjpg-streamer/mjpg-streamer-experimental $ make
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMAKE_BUILD_TYPE = Release
-- Looking for include file sys/inotify.h
-- Looking for include file sys/inotify.h - found
-- Found OpenCV: /usr (found version "3.2.0") found components:  core imgproc highgui videoio 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'libgphoto2'
--   Found libgphoto2, version 2.5.22
-- Found GPHOTO2: /usr/lib/arm-linux-gnueabihf/libgphoto2.so  
-- Looking for include file linux/videodev2.h
-- Looking for include file linux/videodev2.h - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found SDL: /usr/lib/arm-linux-gnueabihf/libSDLmain.a;/usr/lib/arm-linux-gnueabihf/libSDL.so;-lpthread (found version "1.2.15") 
-- Found protobuf-c: /usr/lib/arm-linux-gnueabihf/libprotobuf-c.so  
-- The following features have been enabled:

 * PLUGIN_INPUT_FILE, File input plugin
[... 224 lines cut ...]
-- Installing: /usr/local/share/mjpg-streamer/www/javascript_simple.html
-- Installing: /usr/local/share/mjpg-streamer/www/style.css
-- Installing: /usr/local/share/mjpg-streamer/www/spinbtn_updn.gif
-- Installing: /usr/local/share/mjpg-streamer/www/static.html
-- Installing: /usr/local/share/mjpg-streamer/www/cambozola.jar
-- Installing: /usr/local/share/mjpg-streamer/www/favicon.png
-- Installing: /usr/local/share/mjpg-streamer/www/jquery.rotate.js
-- Installing: /usr/local/share/mjpg-streamer/www/jquery.ui.core.min.js
-- Installing: /usr/local/share/mjpg-streamer/www/functions.js
-- Installing: /usr/local/share/mjpg-streamer/www/java_control.html
-- Installing: /usr/local/share/mjpg-streamer/www/bodybg.gif
-- Installing: /usr/local/share/mjpg-streamer/www/favicon.ico
-- Installing: /usr/local/lib/mjpg-streamer/input_file.so
-- Installing: /usr/local/lib/mjpg-streamer/input_http.so
-- Installing: /usr/local/lib/mjpg-streamer/input_raspicam.so
-- Set runtime path of "/usr/local/lib/mjpg-streamer/input_raspicam.so" to ""
-- Installing: /usr/local/lib/mjpg-streamer/input_ptp2.so
-- Installing: /usr/local/lib/mjpg-streamer/input_uvc.so
-- Installing: /usr/local/lib/mjpg-streamer/output_file.so
-- Installing: /usr/local/lib/mjpg-streamer/output_http.so
-- Installing: /usr/local/lib/mjpg-streamer/output_rtsp.so
-- Installing: /usr/local/lib/mjpg-streamer/output_udp.so
-- Installing: /usr/local/lib/mjpg-streamer/output_viewer.so
make[1]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
```

jacksonliam, 2020-12-15:
Hey, it was written for OpenCV 3.10 so maybe try that version?

It is looking for OpenCV COMPONENTS core imgproc highgui videoio

So make sure they are all included too

ansarid, 2020-12-16:
@jacksonliam I ran `sudo apt install libopencv-core-dev libopencv-videoio-dev libopencv-highgui-dev libopencv-imgproc-dev` and all those packages are already installed but they are for opencv version 3.2.0. I'm having a difficult time finding a package for opencv 3.1.*. I'll see if I can figure that part out though.

luxdong, 2021-01-10:
I can't get de input_opencv.so too. I make for opencv4.11, it that problem?  it work only in opencv 3.10?

pecheronza, 2021-02-07:
I'm having exactly the same issue as OP, and I can't figure out how to install version 3.1.
Any news or workarounds?
Thanks!

j0e1an, 2021-03-03:
I found the solution. Please see the workaround below, cannot guarantee it will not cause any problems since mjp-streamer is written for OpenCV 3.X. 

Edit file `plugins/input_opencv/CMakeLists.txt`
Change `MJPG_STREAMER_PLUGIN_OPTION(input_opencv "OpenCV input plugin"
                            ONLYIF OpenCV_FOUND ${OpenCV_VERSION_MAJOR} GREATER_EQUAL 3)`
To `MJPG_STREAMER_PLUGIN_OPTION(input_opencv "OpenCV input plugin")`

Similarly, edit file `plugins/input_opencv/CMakeLists.txt/filters/cvfilter_py/CMakeLists.txt` and remove the logics of check if OpenCV major version number EQUALS 3. `ONLYIF OpenCV_FOUND ${OpenCV_VERSION_MAJOR} GREATER_EQUAL 3)`

make again and you should see it is enabled.

j0e1an, 2021-03-03:
I notice that OP might have an older version of OpenCV installed, and also pip installed OpenCV. This could cause conflicts. Try to remove the older copy and make again.

ansarid, 2021-06-16:
@j0e1an I finally had the time to test out this solution and I was able to get my OpenCV plugin successfully working without any issues! Thank you!