jacksonliam/mjpg-streamer #283: Unable to use input_http and input_uvc simultaneously (flashing)

MarcelRobitaille, 2020-12-16:
I have two raspberry pis. Both have mjpg-streamer installed. One has two webcams, and the other has one. I want the pi with one webcam to stream data to the other, and the pi with two webcams to proxy that feed, as well as the feeds of its own two webcams.

TLDR: I want this
```
./mjpg_streamer -i "input_uvc.so --device /dev/video0" -i "input_uvc.so --device /dev/video2" -i "input_http.so --host 192.168.50.41" 
```

**Expected behaviour**:
I expect to be able to access the first local webcam at `?action=stream_0`, the second at `?action=stream_1`, and the third at `?action=stream_2`.

**Actual behaviour**:
Both the first webcam and the proxied http input are interlaced in `?action=stream_0`. It keeps flashing between the two feeds.
I get one frame from one, then one frame from the other.

`?action=stream_1` works as expected.

`?action=stream_2` gives a blank screen. Like the other feeds in the browser, the requests never finished, but the screen is always blank.

MarcelRobitaille, 2020-12-16:
Putting `input_http` first in the command seems to be a workaround.

jacksonliam, 2020-12-16:
Interesting use case and bug report, thanks! I guess the input http plugin isn't picking the right buffer to write to!