jacksonliam/mjpg-streamer #284: "./mjpg_streamer: No such file or directory" error after trying to test with ./webcamd

chilicoke, 2020-12-18:
Installed mjpg-streamer on a latest version of bare PI OS Lite, using multiple of install guides including the one in FAQ. Goal is to have a basic and lightweight streamer. 

After making my changes to my /boot/octopi.txt, I used `./webcamd` to make sure everything is working and got the following error:

`pi@Voron22Cam:/root/bin $ ./webcamd
Starting up webcamDaemon...

--- Configuration: ----------------------------
cfg_file:      /boot/octopi.txt
camera:        auto
usb options:   -r 1280x720  -q90 -d /dev/video0
raspi options: -fps 10
http options:  -w ./www -p 8080

Explicitly USB device: /dev/video0
-----------------------------------------------

Found video devices:
/dev/video0
/dev/video1
/dev/video10
/dev/video11
/dev/video12
config file='/boot/octopi.txt':USB device was set in options and found in devices, start MJPG-streamer with the configured USB video device: /dev/video0

Camera model 038f:6001 is known to not work with -f parameter, stripping it out

Camera model 038f:6001 is known to not work with -f parameter, stripping it out

<13>Dec 18 22:41:36 pi: Starting USB webcam
Running ./mjpg_streamer -o output_http.so -w ./www -p 8080 -i input_uvc.so -r 1280x720 -q90 -d /dev/video0
**./webcamd: line 120: ./mjpg_streamer: No such file or directory**
Done bring up all configured video device

Goodbye...`

Any help will be greatly appreciated. 

Thank you

jacksonliam, 2020-12-19:
webcamd is nothing to do with this project.

That being said, it just looks like mjpg-streamer needs to be on your PATH.

closed 2020-12-19 by jacksonliam

Ajimaru, 2020-12-27:
Use the full path to mjpg-streamer in the webcamd.service file.

Example:

`ExecStart=**/home/pi/mjpg-streamer/mjpg-streamer-experimental**/mjpg_streamer -i "**/home/pi/mjpg-streamer/mjpg-streamer-experimental**/input_uvc.so -r 640x480 -f 10 -y -d /dev/video2"  -o "**/home/pi/mjpg-streamer/mjpg-streamer-experimental**/output_http.so -w **/home/pi/mjpg-streamer/mjpg-streamer-experimental**/www -p 8081"`