jacksonliam/mjpg-streamer #285: No input_raspicam.so file after clone

BurningTheSky, 2020-12-26:
I have tried repulling the files but I still don't have an input_raspicam.so file. Am I meant to be getting this from somewhere else or am I just being dense?

`pi@octopi:~ $ git clone https://github.com/jacksonliam/mjpg-streamer.git
Cloning into 'mjpg-streamer'...
remote: Enumerating objects: 5, done.
remote: Counting objects: 100% (5/5), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 2958 (delta 0), reused 1 (delta 0), pack-reused 2953
Receiving objects: 100% (2958/2958), 3.48 MiB | 3.54 MiB/s, done.
Resolving deltas: 100% (1878/1878), done.
pi@octopi:~ $ ls
cron.log  mjpg-streamer  OctoPrint  oprint  scripts
pi@octopi:~ $ cd mjpg-streamer/mjpg-streamer-experimental/
pi@octopi:~/mjpg-streamer/mjpg-streamer-experimental $ ls
cmake  CMakeLists.txt  Dockerfile  docker-start.sh  LICENSE  makedeb.sh  Makefile  mjpg_streamer.c  mjpg_streamer.h  mjpg_streamer@.service  plugins  postinstall.sh  README.md  scripts  start.sh  TODO  utils.c  utils.h  www
pi@octopi:~/mjpg-streamer/mjpg-streamer-experimental $ make
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMAKE_BUILD_TYPE = Release
-- Looking for include file sys/inotify.h
-- Looking for include file sys/inotify.h - found
CMake Warning at plugins/input_opencv/CMakeLists.txt:4 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- Checking for module 'libgphoto2'
--   No package 'libgphoto2' found
-- Could NOT find GPHOTO2 (missing: GPHOTO2_LIBRARY GPHOTO2_INCLUDE_DIR)
-- Looking for include file linux/videodev2.h
-- Looking for include file linux/videodev2.h - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Could NOT find SDL (missing: SDL_LIBRARY SDL_INCLUDE_DIR)
-- Could NOT find protobuf-c (missing: PROTOBUF_C_LIBRARY PROTOBUF_C_INCLUDE_DIR)
-- The following features have been enabled:

 * PLUGIN_INPUT_FILE, File input plugin
 * PLUGIN_INPUT_HTTP, HTTP input proxy plugin
 * PLUGIN_INPUT_UVC, Video 4 Linux input plugin
 * PLUGIN_OUTPUT_FILE, File output plugin
 * PLUGIN_OUTPUT_HTTP, HTTP server output plugin
 * PLUGIN_OUTPUT_RTSP, RTSP output plugin
 * PLUGIN_OUTPUT_UDP, UDP output stream plugin
[... 154 lines cut ...]
-- Installing: /usr/local/share/mjpg-streamer/www/static_simple.html
-- Installing: /usr/local/share/mjpg-streamer/www/style.css
-- Installing: /usr/local/share/mjpg-streamer/www/JQuerySpinBtn.js
-- Installing: /usr/local/share/mjpg-streamer/www/fix.css
-- Installing: /usr/local/share/mjpg-streamer/www/java_simple.html
-- Installing: /usr/local/share/mjpg-streamer/www/jquery.ui.widget.min.js
-- Installing: /usr/local/share/mjpg-streamer/www/rotateicons.png
-- Installing: /usr/local/share/mjpg-streamer/www/jquery.ui.core.min.js
-- Installing: /usr/local/share/mjpg-streamer/www/javascript_simple.html
-- Installing: /usr/local/share/mjpg-streamer/www/jquery.ui.custom.css
-- Installing: /usr/local/share/mjpg-streamer/www/functions.js
-- Installing: /usr/local/share/mjpg-streamer/www/javascript.html
-- Installing: /usr/local/share/mjpg-streamer/www/favicon.png
-- Installing: /usr/local/lib/mjpg-streamer/input_file.so
-- Installing: /usr/local/lib/mjpg-streamer/input_http.so
-- Installing: /usr/local/lib/mjpg-streamer/input_uvc.so
-- Installing: /usr/local/lib/mjpg-streamer/output_file.so
-- Installing: /usr/local/lib/mjpg-streamer/output_http.so
-- Installing: /usr/local/lib/mjpg-streamer/output_rtsp.so
-- Installing: /usr/local/lib/mjpg-streamer/output_udp.so
make[1]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
pi@octopi:~/mjpg-streamer/mjpg-streamer-experimental $ ls
_build  CMakeLists.txt  docker-start.sh  input_http.so  LICENSE     Makefile       mjpg_streamer.c  mjpg_streamer@.service  output_http.so  output_udp.so  postinstall.sh  scripts   TODO     utils.h
cmake   Dockerfile      input_file.so    input_uvc.so   makedeb.sh  mjpg_streamer  mjpg_streamer.h  output_file.so          output_rtsp.so  plugins        README.md       start.sh  utils.c  www
`

jacksonliam, 2020-12-26:
It's not being built. From your log:
```
-- The following features have been disabled:

 * WXP_COMPAT, Enable compatibility with WebcamXP
 * PLUGIN_INPUT_OPENCV, OpenCV input plugin (unmet dependencies)
 * PLUGIN_INPUT_RASPICAM, Raspberry Pi input camera plugin (unmet dependencies)
```

I assume that you are trying to build on a Raspberry Pi? If so which OS? And is it 32-bit?

Because the 64 bit OS for the raspberry Pi do not yet have the features required for input_raspicam to work.

BurningTheSky, 2020-12-26:
Ah, I missed that! I'm building it on a Raspberry Pi 4 Model B 2GB. The OS is Octopi so based on Debian GNU/Linux 10 and it's arm64. Ok, does that mean mjpg streamer on a 64bit pi can't use the camera or is there a workaround?

jacksonliam, 2020-12-26:
Yes you can use input_uvc with the raspicam v4l2 driver

jacksonliam, 2020-12-26:
Duplicate #259

closed 2020-12-26 by jacksonliam